microsoft/AirSim #3928: rcp_error with imagecapture request in cpp

Jaeyoung-Lim, 2021-08-03:
## Bug report 

- AirSim Version/#commit:
- UE/Unity version: Unreal 4.25
- autopilot version: Computer vision mode
- OS Version: Ubuntu 18.04

### What's the issue you encountered?

While it is possible to use the `msr::airlib::MultirotorRpcLibClient` for setting poses with `.simSetVehiclePose()`, when requesting images, it fails with a `rpc_error`
```
Waiting for connection - X
Connected!

Client Ver:1 (Min Req:1), Server Ver:1 (Min Req:1)
terminate called after throwing an instance of 'rpc::rpc_error'
  what():  rpc::rpc_error during call
```

### Settings

```
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "PhysicsEngineName":"ExternalPhysicsEngine",
  "SimMode": "ComputerVision"
}

```

### How can the issue be reproduced?

1. Try using the image API in cpp?

### Include full error message in text form
```
Waiting for connection - X
Connected!

Client Ver:1 (Min Req:1), Server Ver:1 (Min Req:1)
terminate called after throwing an instance of 'rpc::rpc_error'
  what():  rpc::rpc_error during call
```

**What's better than filing an issue? Filing a pull request :).**

Jaeyoung-Lim, 2021-08-03:
Also tried using the `HelloDrone` example and fails as the following

```
~/dev/AirSim/HelloDrone/build$ ./HelloDrone 
Waiting for connection - X
Connected!

Client Ver:1 (Min Req:1), Server Ver:1 (Min Req:1)
Press Enter to get FPV image

Exception raised by the API, something went wrong.
rpclib: function 'simGetImages' (called with 3 arg(s)) threw an exception. The exception is not derived from std::exception. No further information available.
```

jonyMarino, 2021-08-03:
Hi Jaeyoung! You are using the ComputerVision mode but I think what you want is the Multirotor one. Your code should work changing that.

Jaeyoung-Lim, 2021-08-03:
@jonyMarino Thanks for the reply!

How can I capture images when I am in computer vision mode?

 I am trying to use airlib so that it can orient the camera as it is attached on a gimbal, and capture the images through mavlink. Would there be a better way to achieve this without using computer vision mode?

zimmy87, 2021-08-03:
Hi @Jaeyoung-Lim, for capturing images in computer vision mode, I think you'll want to use AirSim's RpcLibClientBase instead of MultirotorRpcLibClient. Your call to simGetImages() should be fine, I think you're just using the wrong client.

jonyMarino, 2021-08-03:
But if you want to use the ROS wrapper, maybe it is better to use the ExternalPhysicsEngine in Multirotor mode. If you have problems with that, let us know.

Jaeyoung-Lim, 2021-08-04:
@zimmy87 Thanks for the inputs, 

I tried using `RpcLibClientBase` as the client, but the `simGetImages()` and `simGetImage()` are both failing with the same errors

```
Waiting for connection - X
Connected!

Client Ver:1 (Min Req:1), Server Ver:1 (Min Req:1)
terminate called after throwing an instance of 'rpc::rpc_error'
  what():  rpc::rpc_error during call
```

Not sure what I am missing here. What I am basically trying to do using airlib. 
 - Set an arbitrary pose of the camera
 - Get the image out of the camera

jonyMarino, 2021-08-04:
Jaeyoung. Neither Zimmy nor I were able to reproduce your issue. Are you using the same AirSim version for the airsim client (HelloDrone) and server (your UE AirSim project)?

Jaeyoung-Lim, 2021-08-05:
@jonyMarino Oh, I forgot to mention this. I built my airlib with gcc-8 and using a landscape mountains binary 1.5.0-linux release.

Do you think this might be causing the issue? Since I am building my package with gcc and linking airlib I thought this was what I should do

Jaeyoung-Lim, 2021-08-05:
By repackaging the latest Airsim on linux solved the problem. (See the image captured below)

[image]

This means that the `1.5.0-linux` release is not compatible anymore with the latest Airsim.

Since I am interested in using the gcc compiled Airlib I tried to build `1.5.0` on gcc-8 but the build script caused a lot of problems.

@jonyMarino @zimmy87 Since my problems were caused by a regression on the airlib API, it would be very helpful if there can be a new linux release, so other people do not need to repackage their Airsim with UE4. Would this be something possible?

Thanks!

jonyMarino, 2021-08-05:
@Jaeyoung-Lim That is in our plans. @saihv is  solving one more issue and then we will start the new release.

Jaeyoung-Lim, 2021-08-05:
@jonyMarino @saihv Any chance of a ETA? would be quite helpful to plan my work :smile:

rajat2004, 2021-08-06:
@Jaeyoung-Lim Re the RPC error with 1.5.0 binary and latest master for compiling the client code, that's mostly due to https://github.com/microsoft/AirSim/pull/3320, I should have thought about and mentioned in the PR at least that it would break binary compatibility with the releases, sorry for that, and something to note at least in the future

jonyMarino, 2021-08-10:
> @jonyMarino @saihv Any chance of a ETA? would be quite helpful to plan my work 😄

This month!

closed 2021-11-04 by Jaeyoung-Lim