microsoft/AirSim #3931: Multi Vehicle in Computer Vision Mode

ale99viv, 2021-08-05:
## Question
### What's your question?

I'm trying to spawn multi computer vision pawn at the start and then control them through python, but when i press start Unreal Engine crash whit this error:
"Fatal error: [File:/home/unreal/jarjar/UnrealVersion/UnrealEngine-4.26/Engine/Source/Runtime/Engine/Private/LevelActor.cpp] [Line: 417] An actor of name 'camera_front_center' already exists in level 'Level 

/Game/UEDPIE_0_MultipleVehicle.MultipleVehicle:PersistentLevel'. << callstack too long >>"

I'm using this settings here:

{
 "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
 "SettingsVersion": 1.2,
 "SimMode": "ComputerVision",
 "PawnPaths": {
  "1": {
   "PawnBP": "Class'/AirSim/Blueprints/CVpawn.CVpawn_C'"
  },
  "2": {"PawnBP": "Class'/AirSim/Blueprints/BP_ComputerVisionPawn.BP_ComputerVisionPawn_C'"
  }
 },
 "Vehicles": {
  "machine_1": {
   "PawnPath": "1",
   "VehicleType": "ComputerVision",
   "X": 20,
   "Y": 0,
   "Z": -2
  },
  "machine_2": {
   "PawnPath": "2",
   "VehicleType": "ComputerVision",
   "X": 70,
   "Y": 0,
   "Z": -2
  }
 }
}

I've already try to duplicate the original computer vision pawn and change its name as well as not specifying PawnPath but the error is the same. 
Someone can help me?

ale99viv, 2021-08-09:
Some help?

jonyMarino, 2021-08-10:
Hi @ale99viv! I'm not sure if this wouldn't be a feature request.
Is using the multirotor mode with the ExternalPhysicsEngine suitable for you?

ale99viv, 2021-08-10:
Hi @jonyMarino, thanks for reply!
Yes could be, but I have not idea how to do this. There are some documentation about this?

jonyMarino, 2021-08-10:
Not much. The idea is that you control the drone with setVehiclePose()

ale99viv, 2021-08-10:
But, if I use setVehiclePose(), the default physic doesn't interfere? For example if i set the height to 50, the drone not fall down?

jonyMarino, 2021-08-11:
No. The ExternalPhysicsEngine  actually disables the physics updates and only "listens" to simSetVehiclePose()

ale99viv, 2021-08-12:
@jonyMarino I've tried and if I set pose with simSetVehiclePose() the drone fall down.
I'm using this simple script to try:
import airsim

client = airsim.MultirotorClient()
client.confirmConnection()

camera = airsim.Pose(airsim.Vector3r(10, 10, -10), airsim.to_quaternion(0, 0, 0))

client.simSetVehiclePose(camera, True, "Drone1")

If you want i can post a short video
Thanks for your replies!!

jonyMarino, 2021-08-12:
Can You share your settings.json file please?
Also, which AirSim version are you using?

ale99viv, 2021-08-12:
I'm using the latest version of AirSim, so 1.5
The settings are this:

{
    "SettingsVersion": 1.2,
    "SimMode": "Multirotor",

    "Vehicles": {
        "Drone1": {
            "VehicleType": "SimpleFlight",
            "X": 0,
            "Y": 0,
            "Z": 0
        }
    }
}

ale99viv, 2021-08-17:
Thanks a lot @jonyMarino ! This work perfectly

closed 2021-08-17 by jonyMarino