microsoft/AirSim #3950: Function 'simGetImages' was called with an invalid number of arguments.

pvhu2002, 2021-08-11:
## Question
### What's your question?

When I try to get images with 'simGetImages',it failed with"Function 'simGetImages' was called with an invalid number of arguments." rpc_error.
### Include context on what you are trying to achieve

N/A
#### Context details

- windows 10
- airsim 1.5.0
- python 3.9.6

- settings.json
https://gist.github.com/pvhu2002/5b8ebe28f8568e6f96604ef4a186c7e1

My code:
responses = client.simGetImages([
    airsim.ImageRequest("0", airsim.ImageType.DepthVis),  #depth visualization image
    airsim.ImageRequest("1", airsim.ImageType.DepthPerspective, True), #depth in perspective projection
    airsim.ImageRequest("1", airsim.ImageType.Scene), #scene vision image in png format
    airsim.ImageRequest("1", airsim.ImageType.Scene, False, False)])  #scene vision image in uncompressed RGBA array
### Include details of what you already did to find answers

rajat2004, 2021-08-11:
Are you using the latest master Pythonclient? In that case, it's likely due to https://github.com/microsoft/AirSim/pull/3320 in combination with 1.5.0 release. You could use the 1.5.0 airsim PyPi package, and remove the `import setup_path` line present in any example files

pvhu2002, 2021-08-12:
> Are you using the latest master Pythonclient? In that case, it's likely due to #3320 in combination with 1.5.0 release. You could use the 1.5.0 airsim PyPi package, and remove the `import setup_path` line present in any example files

Thanks!
I reinstall airsim with 1.5.0 airsim PyPi package and camera api works now.

closed 2021-08-12 by jonyMarino