microsoft/AirSim #3960: Spatial-Temporal Wind Simulation

a-tab-a, 2021-08-12:
## What feature are you suggesting?
### Overview:

I am looking to implement spatial-temporal wind into the simulator, nothing visually but to physics. Ideally I am planning to use wind data from LES simulation. Initially I want to have separate files at different time interval and load the files. Get the drone position, locate the grid in the file, possibly do interpolation depends on the position of drone in grid and then use that wind velocity to calculate drag. What would be the best way to do this? I do want to use it in the SITL too. I assume the wind need to be specified in the settings file. I am thinking adding option like constant or variable wind that will let the simulator pull up file while choosing variable. Tracking from [here](https://github.com/microsoft/AirSim/pull/2867), I see the files needed to update for adding the wind and that's for constant. Could you please help me to get started on this?
### Smaller Details: 

### Nature of Request:  

- Addition
- Change
- Removal
  - 

## Why would this feature be useful?

jonyMarino, 2021-08-12:
Hi @a-tab-a! I don't see a need for a feature request. What you can do is to use the python client, read from a file, and then use the [wind API](https://microsoft.github.io/AirSim/apis/#wind-api)

a-tab-a, 2021-08-12:
Thanks @jonyMarino for replying! I did try that! Unfortunately it will not produce the setup I require, I could be wrong, but it seems the controller does not account for the wind dynamically. For example, using python client,  I set some wind velocity and then for some forward flight I use command self.client.moveToPositionAsync(x, y z, some_velocity) , it moves the drone to certain position and all this time the wind is constant at the value that I set, right? I am looking to change that wind throughout the movement to reach the assigned position, the wind velocity will change depending on the current position of drone and time. I could not achieve that by setting the value once.  Yes, after reaching the position I could again set the wind in the next line but I was looking to adding wind dynamically during that controller does it job, is that doable with the current API? Any suggestion to get started? Looking to test turbulent wind effect. Thanks!

jonyMarino, 2021-08-12:
yes, that is the behavior of moveToPositionAsync.
You can, for example, have a second script that reads the position of the drone and sets the wind accordingly.
Use the latest master not the latest version in the releases because I think it has an issue with setting the wind.

rajat2004, 2021-08-13:
As @jonyMarino mentioned, you could have 2 separate threads and keep following the drone position and updating the wind. It might be somewhat slow and not work properly if the drone is moving quickly, due to the API overheads and delay in client-side processing also. This also wouldn't work if you want to work with multiple vehicles. But this seems like a very specific requirement, and probably not something which will be added to the main repo, but that's up to the maintainers

If you require some assistance, will be happy to help, you could post about issues in modifying the codebase, can discuss about different approaches also. 
One way I can think of for implementing inside the PhysicsEngine itself can be to have a mapping of areas (assuming rectangular, pair of GPS coords) -> wind vector. In each physics update, check which area it lies in, and use that specific wind profile. For better performance, keep track of (cache) the current area instead of searching for it everytime, and only if the vehicle goes out of it, search all the areas and update the new area. For POC of such an initial implementation, you could hard-code the map, but you probably would like to read it from a file rather than rebuilding everytime, this could be added in the `settings.json` also, or if needed could add an API which allows updating the map and read from a different Python script or something and just update the values at runtime as needed apart from the initial file loading (file read can be just removed in this case also, only API can be sufficient)

a-tab-a, 2021-08-16:
Thank you @rajat2004 for replying! It helps a lot! I do require some help doing this, will post as I go!

Zartris, 2021-08-20:
@jonyMarino, what is wrong with the latest release?
I am doing something similar and would love to know if it is my code that is not functioning correctly or if it might be the case of airsim doing something funny :) 

In my case, we think the drone is stabilizing too well, so we are trying to implement "wakes" of wind to push it at some times and we are using the wind API to do the pushes.

jonyMarino, 2021-08-20:
@Zartris  It was a blocking issue. So if it doesn't present to you don't worry. If You want to use current Master commit we are going to use it for releasing version 1.6

Zartris, 2021-08-20:
@jonyMarino: Ahh I have no such problem ;) Thanks for the quick answer

FVT34U, 2025-06-17:
For lost souls, who looking for answers:
```Cast<ASimModeWorldMultiRotor>(UGameplayStatics::GetActorOfClass(World, ASimModeWorldMultiRotor::StaticClass()))->setWind(msr::airlib::Vector3r(NewWind.X, NewWind.Y, NewWind.Z));```
Do this in some class with ```World``` or send it to function with this code, it will set a new vector to ```physics_engine```