microsoft/AirSim #3963: ./build.sh cmake error

EdwinMeriaux, 2021-08-15:
Hi!

So I installed Airsim on a new computer recently and I keep getting the following error when I am working on the rotor params of the drone and using build.sh:

+ debug=false
+ gcc=false
+ [[ 0 -gt 0 ]]
+ RPC_VERSION_FOLDER=rpclib-2.3.0
+ '[' '!' -d ./external/rpclib/rpclib-2.3.0 ']'
+ '[' -d ./cmake_build ']'
++ which cmake
+ CMAKE=/usr/bin/cmake
+ false
+ build_dir=build_release
++ uname
+ '[' Linux == Darwin ']'
+ false
+ export CC=clang-8
+ CC=clang-8
+ export CXX=clang++-8
+ CXX=clang++-8
+ [[ -d ./AirLib/deps/eigen3/Eigen ]]
+ echo 'putting build in build_release folder, to clean, just delete the directory...'
putting build in build_release folder, to clean, just delete the directory...
+ [[ -f ./cmake/CMakeCache.txt ]]
+ [[ -d ./cmake/CMakeFiles ]]
+ [[ ! -d build_release ]]
+ mkdir -p build_release
+ pushd build_release
+ false
+ folder_name=Release
+ /usr/bin/cmake ../cmake -DCMAKE_BUILD_TYPE=Release
-- The C compiler identification is Clang 8.0.0
-- The CXX compiler identification is Clang 8.0.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/clang-8 - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/clang++-8 - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Performing Test UNUSED_LAMBDA_CAPTURE_WARN_SUPPORTED
-- Performing Test UNUSED_LAMBDA_CAPTURE_WARN_SUPPORTED - Success
-- Performing Test ZERO_AS_NULL_POINTER_CONSTANT_WARN_SUPPORTED
-- Performing Test ZERO_AS_NULL_POINTER_CONSTANT_WARN_SUPPORTED - Success
-- Configuring done
CMake Error at AirLib/CMakeLists.txt:26 (add_library):
  No SOURCES given to target: AirLib

CMake Generate step failed.  Build files cannot be regenerated correctly.
+ popd
~/AirSim ~/AirSim
+ rm -r build_release
+ exit 1

So even once I install Airsim, If I go in the current project I am working on, ./update_to_git.sh and then I try to rebuild Airsim I get that error

Anyone know why? I've never seen this before

zimmy87, 2022-01-14:
Are you still experiencing this issue @EdwinMeriaux? Also did you run ./setup.sh before ./build.sh?

thejeshk, 2022-02-17:
Hi, 

I am facing the same issue here - 

ubuntu@ubuntu-OMEN-by-HP-Desktop-PC:~/AirSim$ ./build.sh 
+ debug=false
+ gcc=false
+ [[ 0 -gt 0 ]]
+ RPC_VERSION_FOLDER=rpclib-2.3.0
+ '[' '!' -d ./external/rpclib/rpclib-2.3.0 ']'
+ '[' -d ./cmake_build ']'
++ which cmake
+ CMAKE=/usr/bin/cmake
+ false
+ build_dir=build_release
++ uname
+ '[' Linux == Darwin ']'
+ false
+ export CC=clang-8
+ CC=clang-8
+ export CXX=clang++-8
+ CXX=clang++-8
+ [[ -d ./AirLib/deps/eigen3/Eigen ]]
+ echo 'putting build in build_release folder, to clean, just delete the directory...'
putting build in build_release folder, to clean, just delete the directory...
+ [[ -f ./cmake/CMakeCache.txt ]]
+ [[ -d ./cmake/CMakeFiles ]]
+ [[ ! -d build_release ]]
+ mkdir -p build_release
+ pushd build_release
+ false
+ folder_name=Release
+ /usr/bin/cmake ../cmake -DCMAKE_BUILD_TYPE=Release
-- The C compiler identification is Clang 8.0.1
-- The CXX compiler identification is Clang 8.0.1
-- Check for working C compiler: /usr/bin/clang-8
-- Check for working C compiler: /usr/bin/clang-8 -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/clang++-8
-- Check for working CXX compiler: /usr/bin/clang++-8 -- broken
CMake Error at /usr/share/cmake-3.16/Modules/CMakeTestCXXCompiler.cmake:53 (message):
  The C++ compiler

    "/usr/bin/clang++-8"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/ubuntu/AirSim/build_release/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/usr/bin/make cmTC_58868/fast && /usr/bin/make -f CMakeFiles/cmTC_58868.dir/build.make CMakeFiles/cmTC_58868.dir/build
    make[1]: Entering directory '/home/ubuntu/AirSim/build_release/CMakeFiles/CMakeTmp'
    Building CXX object CMakeFiles/cmTC_58868.dir/testCXXCompiler.cxx.o
    /usr/bin/clang++-8     -o CMakeFiles/cmTC_58868.dir/testCXXCompiler.cxx.o -c /home/ubuntu/AirSim/build_release/CMakeFiles/CMakeTmp/testCXXCompiler.cxx
    Linking CXX executable cmTC_58868
    /usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_58868.dir/link.txt --verbose=1
    /usr/bin/clang++-8       CMakeFiles/cmTC_58868.dir/testCXXCompiler.cxx.o  -o cmTC_58868 
    /usr/bin/ld: cannot find -lstdc++
    clang: error: linker command failed with exit code 1 (use -v to see invocation)
    make[1]: *** [CMakeFiles/cmTC_58868.dir/build.make:87: cmTC_58868] Error 1
    make[1]: Leaving directory '/home/ubuntu/AirSim/build_release/CMakeFiles/CMakeTmp'
    make: *** [Makefile:121: cmTC_58868/fast] Error 2
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:2 (project)

-- Configuring incomplete, errors occurred!
See also "/home/ubuntu/AirSim/build_release/CMakeFiles/CMakeOutput.log".
See also "/home/ubuntu/AirSim/build_release/CMakeFiles/CMakeError.log".
+ popd
~/AirSim ~/AirSim
+ rm -r build_release
+ exit 1