microsoft/AirSim #3971: The Mac Instructions with the opening Blocks.xcworkspace doesn't build nor run so I cannot run the simulator with the HelloCar project.

shyamalschandra, 2021-08-19:
## Bug report 
- AirSim Version/#commit: 19fea1eac55531c5d5d6ebd91c8101f330f3e549
- UE/Unity version: 5.0.0 Early Access 2
- autopilot version: N/A
- OS Version: macOS Big Sur 11.5.2

### What's the issue you encountered?

The Blocks.xcworkspace doesn't work, build, nor run.

[image]

### Settings
N/A

### How can the issue be reproduced?
Following the directions for MacOS for AirSim.

### Include full error message in text form

I already included the number of errors in Xcode.

rajat2004, 2021-08-20:
I don't think 5.0 will work with AirSim currently, PhysX has been removed so the Car vehicle will need to be adapted to either the Chaos engine or maybe the FastPhysicsEngine if someone's up to it

zimmy87, 2022-01-14:
Hey @shyamalschandra are you still experiencing this issue?