microsoft/AirSim #3972: rpclib: client error C0002: Function 'simGetImages' was called with an invalid number of arguments. Expected: 2, got: 3

digiwang, 2021-08-19:
## Bug report 

- AirSim Version/#commit: **1.5.0/19fea1eac55531c5d5d6ebd91c8101f330f3e549**
- UE/Unity version: **UE 4.26.1**
- autopilot version: **N/A**
- OS Version: **Ubuntu 18.04**

### What's the issue you encountered?

I was trying to see if my current configuration was working properly.  I can confirm that a previous setup with Airsim v1.5.0 and UE v4.26.1 was working.

### Settings

```
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "ViewMode": "SpringArmChase",
  "ClockSpeed": 1.0,
  "Vehicles": {
    "drone_1": {
      "VehicleType": "SimpleFlight",
      "DefaultVehicleState": "Armed",
      "EnableCollisionPassthrogh": false,
      "EnableCollisions": true,
      "AllowAPIAlways": true,
      "RC": {
        "RemoteControlID": 0,
        "AllowAPIWhenDisconnected": false
      },
      "Cameras": {
        "front_center_custom": {
          "CaptureSettings": [
            {
              "PublishToRos": 1,
              "ImageType": 0,
              "Width": 640,
              "Height": 480,
              "FOV_Degrees": 27,
              "DepthOfFieldFstop": 2.8,
              "DepthOfFieldFocalDistance": 200.0, 
              "DepthOfFieldFocalRegion": 200.0,
              "TargetGamma": 1.5
            }
          ],
          "X": 0.50, "Y": 0, "Z": 0.10,
          "Pitch": 0, "Roll": 0, "Yaw": 0
        },
        "front_left_custom": {
          "CaptureSettings": [
            {
              "PublishToRos": 1,
              "ImageType": 0,
              "Width": 672,
              "Height": 376,
              "FOV_Degrees": 90,
              "TargetGamma": 1.5
            }, 
            {
              "PublishToRos": 1,
              "ImageType": 2,
              "Width": 672,
              "Height": 376,
              "FOV_Degrees": 90,
              "TargetGamma": 1.5
            }
          ],
          "X": 0.50, "Y": -0.06, "Z": 0.10,
          "Pitch": 0.0, "Roll": 0.0, "Yaw": 0.0
        },
        "front_right_custom": {
          "CaptureSettings": [
            {
              "PublishToRos": 1,
              "ImageType": 0,
              "Width": 672,
              "Height": 376,
              "FOV_Degrees": 90,
              "TargetGamma": 1.5
            }
          ],
          "X": 0.50, "Y": 0.06, "Z": 0.10,
          "Pitch": 0.0, "Roll": 0.0, "Yaw": 0.0
        }
      },
      "X": 2, "Y": 0, "Z": 0,
      "Pitch": 0, "Roll": 0, "Yaw": 0
    }
  },
  "SubWindows": [
    {"WindowID": 0, "ImageType": 0, "CameraName": "front_left_custom", "Visible": true},
    {"WindowID": 1, "ImageType": 0, "CameraName": "front_center_custom", "Visible": false},
    {"WindowID": 2, "ImageType": 0, "CameraName": "front_right_custom", "Visible": true}
  ]
}
```

### How can the issue be reproduced?

```
import airsim
import cv2
import numpy as np

c = airsim.MultirotorClient()

c.confirmConnection()

responses = c.simGetImages([airsim.ImageRequest("fron_center", airsim.ImageType.DepthPlanar, True)])
response = responses[0]
img1d = response.image_data_float
img2d = np.reshape(img1d, (response.height, response.width))
cv2.imshow("Depth viz", img2d)
cv2.waitKey(0)
```

### Include full error message in text form
Exception has occurred: RPCError
rpclib: client error C0002: Function 'simGetImages' was called with an invalid number of arguments. Expected: 2, got: 3
  File "/tmp/test/test.py", line 9, in <module>
    responses = c.simGetImages([airsim.ImageRequest("fron_center", airsim.ImageType.DepthPlanar, True)])

**What's better than filing an issue? Filing a pull request :).**

lyvelyte, 2021-08-19:
I am also having the exact same issue. This is what prompted to make a post in the discussion board requesting software versions because I suspected it may be a library version issue. 

I am using Windows 10 x64 with python 3.7.11. I've tried UE 4.25 and 4.26.

digiwang, 2021-08-19:
@lyvelyte, I suspect the same thing. FYI, I updated my initial post to include information that, *keeping all else fixed*, I also tried running the above code against the release versions of UE v4.26.2 and v4.26.0, but still no go for me. Unfortunately, I cannot go back to UE v4.25.0 as my custom environment uses functionality that was introduced in UE v4.26.0 or I would test with that too.

I am going to go back to UE v4.26.1 and revert my AirSim library version to v1.4.0. I'll post my findings when done.

jonyMarino, 2021-08-19:
Hi! this is a dup of #3950. You need to use the same version for the AiSim client and server.

digiwang, 2021-08-19:
@jonyMarino, omg... I could have sworn I checked that, but obviously there was a flaw in my workflow because now the issue is resolved. Thank you so much!

**My New Workflow:**
1. Build UE
2. In Airsim (server) root repo folder, run `git checkout <tag>`
3. In Airsim (server) root repo folder, run `clean_rebuild.sh`
4. In Airsim (server) root repo folder, run `rsync -a --delete Unreal/Plugins path/to/MyUnrealProject`
5. Check Airsim (client) version, `python3 -m pip show airsim` and install the correct version that matches the server version.
6. Start UE4Editor, open *MyUnrealProject* and rebuild AirSim plugin when asked.
7. **cross fingers**

closed 2021-08-19 by jonyMarino

matsuyoi2022, 2023-04-06:
Thanks a lot. My Airsim version is 1.8.1, but the version of airsim python library is 1.5.0, then I meet the same mistake.

matsuyoi2022, 2023-04-06:
> Hi! this is a dup of #3950. You need to use the same version for the AiSim client and server.

yes that's it

978396637, 2023-07-20:
I fix this problem, just make sure that your server and client build from the same Airsim Project. In addition, the version airsim python library should be the same.