microsoft/AirSim #3993: Cannot use AirSim camera API

BrenoPimentel, 2021-08-28:
I'm using AirSim to my drone simulations. Installed PX4 and after that, my scripts does not work properly. Can't even enable API control, but what i really need use is AirSim camera API, and it is not working.
Get error trying to run the following code:
`
import airsim
import os
import cv2
import numpy as np

# connect to the AirSim simulator
client = airsim.MultirotorClient()

while True:
    print("connecting")
    responses = client.simGetImages([airsim.ImageRequest(0, airsim.ImageType.Scene, True)])
    response = responses[0]

    # get numpy array
    img1d = np.fromstring(response.image_data_uint8, dtype=np.uint8)

    # reshape array to 4 channel image array H X W X 4
    img_rgb = img1d.reshape(response.height, response.width, 3)

    cv2.imshow('tela', img_rgb)
    k = cv2.waitKey(1)
    #if k == 0:
    #    cv2.closeAllWindows()
`
Got this error:
`
C:\Users\breno\Documents\BlackBeeDrones\AirSim_BlackBee\Codigos_Python_AirSim\venv\Scripts\python.exe C:/Users/breno/Documents/BlackBeeDrones/AirSim_BlackBee/Codigos_Python_AirSim/airsim_cam_test.py
Traceback (most recent call last):
  File "C:\Users\breno\Documents\BlackBeeDrones\AirSim_BlackBee\Codigos_Python_AirSim\airsim_cam_test.py", line 11, in <module>
    responses = client.simGetImages([airsim.ImageRequest(0, airsim.ImageType.Scene, True)])
  File "C:\Users\breno\Documents\BlackBeeDrones\AirSim_BlackBee\Codigos_Python_AirSim\venv\lib\site-packages\airsim\client.py", line 264, in simGetImages
    responses_raw = self.client.call('simGetImages', requests, vehicle_name)
  File "C:\Users\breno\Documents\BlackBeeDrones\AirSim_BlackBee\Codigos_Python_AirSim\venv\lib\site-packages\msgpackrpc\session.py", line 41, in call
connecting
    return self.send_request(method, args).get()
  File "C:\Users\breno\Documents\BlackBeeDrones\AirSim_BlackBee\Codigos_Python_AirSim\venv\lib\site-packages\msgpackrpc\future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: client error C0002: Function 'simGetImages' was called with an invalid number of arguments. Expected: 2, got: 3
`

BrenoPimentel, 2021-08-28:
here is my AirSim settings.json:
`{
    "SettingsVersion": 1.2,
    "SimMode": "Multirotor",
    "ClockType": "SteppableClock",
    "DefaultVehicleConfig": "PX4",
    "Vehicles": {
        "PX4": {
            "VehicleType": "PX4Multirotor",
            "UseSerial": false,
            "LockStep": true,
            "UseTcp": true,
            "TcpPort": 4560,
            "ControlIp": "localhost",
            "ControlPortLocal": 14540,
            "ControlPortRemote": 14580,
            "LocalHostIp": "172.20.0.1",
            "Sensors": {
                "Barometer": {
                    "SensorType": 1,
                    "Enabled": true,
                    "PressureFactorSigma": 0.0001825
                }
            },
            "Parameters": {
                "NAV_RCL_ACT": 0,
                "NAV_DLL_ACT": 0,
                "COM_OBL_ACT": 1,
                "LPE_LAT": 47.641468,
                "LPE_LON": -122.140165
            },
            "CameraDefaults": {
                "CaptureSettings": [
                    {
                        "ImageType": 0,
                        "Width": 320,
                        "Height": 180,
                        "FOV_Degrees": 80
                    },
                    {
                        "ImageType": 3,
                        "Width": 320,
                        "Height": 180,
                        "FOV_Degrees": 80
                    }
                ]
            }
        }
    }
}`

rajat2004, 2021-08-29:
You're probably using the 1.5.0 release binaries with master or 1.5.0 Python package. Try using the 1.6.0 binaries, or if needed 1.5 binaries with 1.5 Python package

BrenoPimentel, 2021-08-29:
Still doesn't work, I have update airsim package to 1.6.0 and downloaded binares 1.6.0, but even doing it, get the same error.

BrenoPimentel, 2021-08-29:
Running airsim version 1.5.0 with 1.5.0 binaries i get the following error:
[image]

curious-energy, 2021-08-30:
Hello, now I meet same thing when using 1.6.0 binaries(linux)，but win version is no problem.

issue in linux binaries: all cameras in airsim_ros_pkg don't work for me.

curious-energy, 2021-08-30:
> Hello, now I meet same thing when using 1.6.0 binaries(linux)，but win version is no problem.
> 
> issue in linux binaries: all cameras in airsim_ros_pkg don't work for me.

**Now I try 1.5.0 binaries(linux), all things is right!**

jonyMarino, 2021-08-30:
Hi @BrenoPimentel, and welcome! What does "pip show airsim" returns to you?

BrenoPimentel, 2021-08-31:
hi, Version 1.6.0

jonyMarino, 2021-09-15:
The error is clearly for using a binary prior to version 1.6... check that please