microsoft/AirSim #3994: msgpackrpc.error.RPCError when running hello_drone.py and calling enableApiControl

abuchnick, 2021-08-29:
## Bug report 

- AirSim Version/#commit: Airsim - 1.5.0
- UE/Unity version: UE - 4.25.4
- autopilot version: N/A
- OS Version: Win 10 Enterprise x64

### What's the issue you encountered?

Hello, I just finished building and installing airsim and want to use the python API for my project.
When trying to run hello_drone.py (which is part of the repo, under PythonClient/multirotor), I get the above mentioned exception. 
I've seen discussions about this problem, but not under this repo, only under "drone racing" repo.
When running into this problem. the UE is up and the VS runs the C++ code (I'm trying to run the airsim python module from PyCharm).

[image]
[image]
[image]

### Settings

### How can the issue be reproduced?

1. Run hello_drone.py
2. Profit

### Include full error message in text form

> msgpackrpc.error.RPCError: rpclib: function 'enableApiControl' (called with 2 arg(s)) threw an exception. The exception contained this information: Vehicle API for '' is not available. This could either because this is simulation-only API or this vehicle does not exist.

**What's better than filing an issue? Filing a pull request :).**

alonfaraj, 2021-08-29:
Has the drone actually been spawned? Can you see it in Unreal?
I get an error using your `settings.json` because of the last comma after `"Multirotor"`

abuchnick, 2021-08-29:
> 
> 
> Has the drone actually been spawned? Can you see it in Unreal?
> I get an error using your `settings.json` because of the last comma after `"Multirotor"`.
> Are you sure this is the json file AirSim loads?

Hey, thanks for your quick reply!
I guess this was a real problem haha, I fixed it and reinstalled & rebuilt the entire project, and now it works.
Thanks!

closed 2021-08-29 by abuchnick