microsoft/AirSim #4011: socket error

LLM1602, 2021-09-05:
## Bug report 

- AirSim Version/#commit:1.5.0-windows
- UE/Unity version:ue_4.26
- autopilot version:drone
- OS Version:windows10

### What's the issue you encountered?

when i run the dqn for drone to follow a random target,i get the error when the total_timesteps is 8100(the number can be others)  

### Settings

<the settings.json>
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor"
}

### How can the issue be reproduced?

1.I just run the dqn code that i used every day,but yesterday i get error when i continuing tarining the model 
2.I think it is not my code's problem,becasue it used every day jsut training and not changed,but yesterday it didn't work studdently

### Include full error message in text form

Traceback (most recent call last):

  File "E:/pycharm_project/AirSim-master/PythonClient/reinforcement_learning/dqn_drone_tracing_random.py", line 65, in <module>

    model.learn(

  File "E:\pycharm_project\test_DRL\airsim_DRL\lib\site-packages\stable_baselines3\dqn\dqn.py", line 240, in learn

    return super(DQN, self).learn(

  File "E:\pycharm_project\test_DRL\airsim_DRL\lib\site-packages\stable_baselines3\common\off_policy_algorithm.py", line 352, in learn

    rollout = self.collect_rollouts(

  File "E:\pycharm_project\test_DRL\airsim_DRL\lib\site-packages\stable_baselines3\common\off_policy_algorithm.py", line 573, in collect_rollouts

    new_obs, reward, done, infos = env.step(action)

  File "E:\pycharm_project\test_DRL\airsim_DRL\lib\site-packages\stable_baselines3\common\vec_env\base_vec_env.py", line 162, in step

    return self.step_wait()

  File "E:\pycharm_project\test_DRL\airsim_DRL\lib\site-packages\stable_baselines3\common\vec_env\vec_transpose.py", line 83, in step_wait

    observations, rewards, dones, infos = self.venv.step_wait()

  File "E:\pycharm_project\test_DRL\airsim_DRL\lib\site-packages\stable_baselines3\common\vec_env\dummy_vec_env.py", line 43, in step_wait

    obs, self.buf_rews[env_idx], self.buf_dones[env_idx], self.buf_infos[env_idx] = self.envs[env_idx].step(

  File "E:\pycharm_project\test_DRL\airsim_DRL\lib\site-packages\stable_baselines3\common\monitor.py", line 90, in step

    observation, reward, done, info = self.env.step(action)

  File "E:\pycharm_project\AirSim-master\PythonClient\reinforcement_learning\airgym\envs\drone_env.py", line 181, in step

    obs = self._get_obs()

  File "E:\pycharm_project\AirSim-master\PythonClient\reinforcement_learning\airgym\envs\drone_env.py", line 88, in _get_obs

    image = self.transform_obs(responses)

  File "E:\pycharm_project\AirSim-master\PythonClient\reinforcement_learning\airgym\envs\drone_env.py", line 81, in transform_obs

    image = Image.fromarray(img2d)

  File "E:\pycharm_project\test_DRL\airsim_DRL\lib\site-packages\PIL\Image.py", line 2793, in fromarray

    return frombuffer(mode, size, obj, "raw", rawmode, 0, 1)

  File "E:\pycharm_project\test_DRL\airsim_DRL\lib\site-packages\PIL\Image.py", line 2733, in frombuffer

    return frombytes(mode, size, data, decoder_name, args)

  File "E:\pycharm_project\test_DRL\airsim_DRL\lib\site-packages\PIL\Image.py", line 2679, in frombytes

    im.frombytes(data, decoder_name, args)

  File "E:\pycharm_project\test_DRL\airsim_DRL\lib\site-packages\PIL\Image.py", line 796, in frombytes

    d.setimage(self.im)

ValueError: tile cannot extend outside image

WARNING:tornado.general:Write error on <socket.socket fd=-1, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0>: [WinError 10038] 在一个非套接字上尝试了一个操作。

Exception ignored in: <function AirSimDroneEnv.__del__ at 0x0000026060FB7EE0>

Traceback (most recent call last):

  File "E:\pycharm_project\AirSim-master\PythonClient\reinforcement_learning\airgym\envs\drone_env.py", line 45, in __del__

  File "E:\pycharm_project\AirSim-master\PythonClient\airsim\client.py", line 26, in reset

  File "E:\pycharm_project\test_DRL\airsim_DRL\lib\site-packages\msgpackrpc\session.py", line 41, in call

  File "E:\pycharm_project\test_DRL\airsim_DRL\lib\site-packages\msgpackrpc\future.py", line 25, in get

  File "E:\pycharm_project\test_DRL\airsim_DRL\lib\site-packages\msgpackrpc\future.py", line 22, in join

  File "E:\pycharm_project\test_DRL\airsim_DRL\lib\site-packages\msgpackrpc\loop.py", line 22, in start

  File "E:\pycharm_project\test_DRL\airsim_DRL\lib\site-packages\tornado\ioloop.py", line 863, in start

  File "E:\pycharm_project\test_DRL\airsim_DRL\lib\site-packages\tornado\platform\select.py", line 62, in poll

OSError: [WinError 10038] 在一个非套接字上尝试了一个操作。

Process finished with exit code 1

**What's better than filing an issue? Filing a pull request :).**

necharte, 2021-12-11:
Were you able to find a solution? I'm having exactly the same problem...

Samuel-Fipps, 2023-08-09:
I am having this issue as well...

Samuel-Fipps, 2023-08-09:
I fixed this by using a older version of stable baseline

BetterLL, 2023-11-19:
> I fixed this by using a older version of stable baseline

I have the same problem, which version of stable baseline do you usse?Thanks.

Samuel-Fipps, 2023-12-14:
stable_baselines3==1.4.0