microsoft/AirSim #4014: If you move the drone to the right or left diagonal (45° or 135°), the drone will rotate slightly in that direction.

RinR0000, 2021-09-06:
If you move the drone to the right or left diagonal (45° or 135°), the drone will rotate slightly in that direction.

Why is this?

PX4(SITL,WSL2)
Python API: moveByVelocityBodyFrame

jonyMarino, 2021-09-13:
Hi @RinR0000! Can You provide the full code to reproduce the issue?

RinR0000, 2021-09-13:
@jonyMarino 
Thanks for the reply.

I'm using Pygames to read the Xbox controller and control the drone.

I'm assigning the joystick tilt to X and Y and passing it to the Python API. ↓

Self.client.
self.client.moveByVelocityZBodyFrameAsync(x*(7),y*7,self.height,0.01,airsim.DrivetrainType.MaxDegreeOfFreedom,airsim.YawMode(is_ rate=True, yaw_or_rate=yaw)).join()
YawMode(is_ rate=True, yaw_or_rate=yaw)).join()

Is the aircraft shaking when moving in the diagonal direction? (In the Yaw direction)

jonyMarino, 2021-10-05:
@RinR0000 , can you try using mavsdk instead of the airsim client?