microsoft/AirSim #4024: Drone falls off when spawning in costume made environment

caesar84, 2021-09-13:
## Question
### I have a question about why my drone falls off when spawning in a costume-made environment whereas replacing the drone with the car works fine.

### Sorry for multiple posting as I dont know if this or the the question section is more appropriate .
I have imported a costume environment of a street in a city to Unreal engine and the settings were ok. tried to run a drone and after hitting play the drone falls off and goes down the road like it doesn't sense the pavement at all. when replacing the drone with the car it never falls even though the player start position is the same. Am I missing something here?. The second issue, my scene is currently far from (0,0,0) therefore every time I stop the simulation the starting point is far away that I have to navigate towards the scene which takes time. Due to the complexity of my scene, Is there any possible way that i change the whole scene position in one go  without the need to move each part alone or a quick way to resume the simulation in a position within the scene rather than the origin (0,0,0)  

#### Context details
<! windows 10, latest airsim version, VS2019, Unreal 26.4,  >
<! {
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode":"Multirotor",
  "Vehicles": {
        "Drone1": {
          "VehicleType": "SimpleFlight",
          "X": -10, "Y": 0, "Z": 0,
      "Yaw": -180
		}
        },
  "CameraDefaults": {
      "CaptureSettings": [
        {
          "ImageType": 0,
          "Width": 512,
          "Height": 400,
          "FOV_Degrees": 90,
          "AutoExposureSpeed": 100,
          "AutoExposureBias": 0,
        "AutoExposureMaxBrightness": 0.64,
        "AutoExposureMinBrightness": 0.03
        }
    ]
  }
}
 >

### Include details of what you already did to find answers

caesar84, 2021-09-14:
@jonyMarino can you please just have a look at this, is it a bug?

jonyMarino, 2021-09-15:
Hi @caesar84! You should check the collision settings for the floor inside the editor. @zimmy87 do you remember the exact settings that are needed?

caesar84, 2021-09-15:
@jonyMarino thank you very much for that note and waiting for @zimmy87 to help , hopefully. Sorry if tagging you was bit inappropriate but I have struggled to sort it out . Should I change the collision settings for the random cars also as the drone fails to detect a collision with them as well?. Finally, does it make sense to you that drone fails while the car detects the collision even it's exactly the same environment? 
Can't thank you enough for your reply .

jonyMarino, 2021-09-15:
No worries.
Take a look at this comment:
https://github.com/microsoft/AirSim/issues/3855#issuecomment-874984693

That was what i was refering to.

caesar84, 2021-09-15:
hi @jonyMarino, thank you very much It turns out ground and many buildings have no cillision. thanks alot.

closed 2021-09-17 by jonyMarino

Kabilan-n, 2023-03-30:
Hi, I'm also facing the same error when I'm trying to use drone i9n custom environment it is falling under the ground and there is no collision with the other buildings also where as when I'm trying with car I was able to get in on the ground and the collisions with buildings are also working properly.

I tried adding collision to the objects and the player (i.e drone) but still it fails to land on the floor. and added { "EnableCollisionPassthrogh": false, "EnableCollisions": true } in my settings.json

This is what I get when I try using car. It was able to land in the ground. 
[image]

When I try using drone the following happens. 
[image]

[image]