microsoft/AirSim #4041: Custom function in client.py report rpc_error

LittleTimeWxk, 2021-09-27:
## Bug Report

- AirSim Version/#commit: 1.5.0
- UE/Unity version: 4.26
- autopilot version:
- OS Version: UE4 in windows10,   hello.py in ubuntu16.04 , ros version is kenetic

### What's the issue you encountered?

Precondition：
    I use UE to create a custom environment in the windows environment, and introduce the AirSim plug-in, and use the ubuntu system as a client to control airsim vehicles under windows. 
    I customized a new function in client.py, hoping to get the name of the current computer through this function and return it to the UE to record. Of course, I also implemented the custom function implementation and rpc binding in the airsim source code on the UE side. But I called a custom function in the confirmConnection function in client.py, and then I reported an error that the server could not find the function.

Logically speaking, there is generally no error in customizing the new API for rpc communication. I find it very strange.

### Settings

﻿{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Car",
  "Map": "pujiang_map.xml",
  "CarType": "SUV",
  "ControlCarCount": 1,
  "ShowVechicleTrajecory":0,
  "EnvCarCount": 0,
  "Vehicles":{
    "SUV":{
      "x": 0,
      "y": 0,
      "z": 0,
      "VehicleType": "physxcar",
      "MaxSteering": 35
    }
  }
}

### How can the issue be reproduced?

1.
2.

### Include full error message in text form
(base) admin@admin:~/codes/autonomous_driving/install/lib/ms_planning_simulation_node$ python hello_car.py 
192.168.2.120
Connected!
Client Ver:1 (Min Req: 1), Server Ver:1 (Min Req: 1)

host name is :xueshan
this computer ip is :127.0.1.1
Traceback (most recent call last):
  File "/home/admin/codes/airsimTestProject/hello_car.py", line 10, in <module>
    client.confirmConnection()
  File "/home/admin/codes/airsimTestProject/airsim/client.py", line 201, in confirmConnection
    self.carname = self.getControllCarName()
  File "/home/admin/codes/airsimTestProject/airsim/client.py", line 214, in getControllCarName
    res = self.client.call('getControllCarName', ip, hostname)
  File "/home/admin/miniconda3/lib/python3.9/site-packages/msgpackrpc/session.py", line 41, in call
    return self.send_request(method, args).get()
  File "/home/admin/miniconda3/lib/python3.9/site-packages/msgpackrpc/future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: server could not find function 'getControllCarName' with argument count 2.

**What's better than filing an issue? Filing a pull request :).**

jonyMarino, 2021-10-07:
Hi @LittleTimeWxk!
See if you are following the steps below after modifying the API code on the simulator side:

call /AirSim/Unreal/Environments/Blocks/update_to_git.sh
call /AirSim/clean.sh
call /AirSim/build.sh
call /AirSim/Unreal/Environments/Blocks/update_from_git.sh

LittleTimeWxk, 2021-10-09:
I solved this problem!
The reason for this problem is that there is a conditional compilation problem in the airsim source code in Visual Studio. You can open the source file called CarRpcLibServer.cpp. The conditionally compiled code and comments at the top of the file.
'''
#ifndef AIRLIB_HEADER_ONLY
//RPC code requires C++14. If build system like Unreal doesn't support it then use compiled binaries
...
#endif
'''
I commented out the code of conditional compilation and recompiled the custom rpc communication code to take effect.

closed 2021-10-09 by LittleTimeWxk