microsoft/AirSim #4054: Eigen Repository was deleted.

SamuelSHernandez, 2021-10-03:
## Bug report 

- AirSim Version/#commit: 2b01754aad58cafa5b68308a9c72482ca6baf8b9
- UE/Unity version: 4.25.4
- autopilot version: N/A
- OS Version: MacOS Big Sur Version 11.6

### What's the issue you encountered?

Eigen3 repository was deleted
I discovered the issue while trying to run './setup.sh' and I encountered this error:

[image]
. 

I tried re-installing the file individually with 'curl -o eigen3.zip https://gitlab.com/libeigen/eigen/-/archive/3.3.7/eigen-3.3.7.zip'

The file downloaded successfully but yielded the same error.

I tried to see if it was a problem with the file type so I checked for its type with 'file eigen3.zip' which yielded 'eigen3.zip: HTML document text, ASCII text, with no line terminators'

And following the link I encountered this.

[image]

I filled an issue in Gitlab 

### How can the issue be reproduced?

```
git clone https://github.com/Microsoft/AirSim.git
cd AirSim
./setup.sh
```

### Include full error message in text form
Saving to: ‘eigen3.zip’

eigen3.zip                            [ <=>                                                       ]  74.39K  --.-KB/s    in 0.02s   

2021-10-03 02:19:25 (3.39 MB/s) - ‘eigen3.zip’ saved [76172]

+ unzip -q eigen3.zip -d temp_eigen
[eigen3.zip]
  End-of-central-directory signature not found.  Either this file is not
  a zipfile, or it constitutes one disk of a multi-part archive.  In the
  latter case the central directory and zipfile comment will be found on
  the last disk(s) of this archive.
unzip:  cannot find zipfile directory in one of eigen3.zip or
        eigen3.zip.zip, and cannot find eigen3.zip.ZIP, period.

**What's better than filing an issue? Filing a pull request :).** 
I wish I could, but problem is out of my control.

skykim0609, 2021-10-04:
I faced the same issue, and bypassed it by copying the eigen3 under /usr/include/ to /AirLib/deps. At least it made the build successful.

jonyMarino, 2021-10-04:
Eigen repository is back again https://gitlab.com/libeigen/eigen/-/issues/2336

closed 2021-10-04 by jonyMarino