microsoft/AirSim #4062: ERROR: new version of AirSim requires newer rpclib.

kulkarnikeerti, 2021-10-05:
I am trying to have the AirSim for Linux version. I have the running UnrealEngine but when I try to setup and build AiSim I end up with below error in this line.

```
echo 'ERROR: new version of AirSim requires newer rpclib.'
ERROR: new version of AirSim requires newer rpclib.
+ echo 'please run setup.sh first and then run build.sh again.'
please run setup.sh first and then run build.sh again.
```

https://github.com/microsoft/AirSim/blob/2b01754aad58cafa5b68308a9c72482ca6baf8b9/build.sh#L35

Any already existing solution for this? or did I miss out something here?

jonyMarino, 2021-10-05:
Hi @kulkarnikeerti and welcome! Have you run setup.sh before build.sh?

kulkarnikeerti, 2021-10-05:
@jonyMarino yes, I did.

jonyMarino, 2021-10-05:
@kulkarnikeerti can you delete your external/rpclib folder and try again?

kulkarnikeerti, 2021-10-05:
@jonyMarino I am really sorry if this seems to be silly. But I really can't find the folder. Could you please help me where can I find it? Is it created after running setup.sh?

kulkarnikeerti, 2021-10-05:
I am not really sure if this makes a difference. Running setup.sh has not created a this folder. And I also seem to see this error when I run setup.sh

```
E: The repository 'http://ppa.launchpad.net/umang/indicator-stickynotes/ubuntu hirsute Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```
Does this make any difference?

jonyMarino, 2021-10-05:
Yes. setup.sh is failing before the rpclib is downloaded. As the error says, you have an issue with the ppa you are using for Ubuntu. Which version of Ubuntu are you using? I'm not an expert on this, I think you can dig more into this error to find a solution. Please keep me updated.

kulkarnikeerti, 2021-10-06:
I am using Ubuntu 21.04. And yes, I will research a bit on this and keep you update. Thanks!

closed 2021-10-08 by kulkarnikeerti

jonyMarino, 2021-10-08:
@kulkarnikeerti  What machine are you using? Do you have a graphics card?

jonyMarino, 2021-10-08:
Maybe a driver issue with Unreal?

debian-user-france1, 2021-12-06:
Do you see this: `LogPackageName: SearchForPackageOnDisk took   0.008s, but failed to resolve Blocks.`?