microsoft/AirSim #4075: How to update AirSim Setting.json

eherozhao, 2021-10-12:
## Question
### What's your question?
I don't know how to update Settings.json, the official documentation is not helpful. When I use 
{
  "SettingsVersion": 1.2,
  "SimMode": "Car",
  "Vehicles": {
    "Drone1": {
       "VehicleType": "SimpleFlight",
        "X": 0.1, "Y": 0, "Z": -3, "Yaw": 0
     },
    "Car1": {
       "VehicleType": "PhysXCar",
       "X": 0, "Y": 0, "Z": -1, "Yaw": 0
     }
  }
}
It still tells me no compatible car.

### Include context on what you are trying to achieve
N/A
#### Context details
I cannot use AirSim 1.6 because of linking error: opencv so I used 1.4 with Unreal Engine 4.26

### Include details of what you already did to find answers

jonyMarino, 2021-10-14:
Hi @eherozhao! You can not mix cars and drones in the same simulation. So using "SimMode": "Car," you can have only cars in the vehicles block.

eherozhao, 2021-10-18:
Is there any way to create the traffic with moving vehicles?

zimmy87, 2021-11-02:
I do not believe we have the proper license for sharing the source of the traffic in the City environment, though it is based upon the Path Follow plugin in Unreal Marketplace: https://www.unrealengine.com/marketplace/en-US/product/path-follow. That would be a good place to start for creating moving vehicles in the environment.

eherozhao, 2022-04-08:
Sorry for the late reply and thank you for the answer. Our team launched an Unreal Engine 4 plugin to procedurally create content for autonomous vehicles and robotics simulation areas. I used AirSim with our plugin and it worked together. https://github.com/aws-samples/aws-ambit-scenario-designer-ue4.