microsoft/AirSim #4078: Packaging issues（windows to Linux packaging）

Hijiongfeng, 2021-10-13:
## Question
### What's your question?

[image]
I am using the windows platform to package to the linux platform and want to generate a file that can be executed under linux. I followed the above figure to operate, but the following problems occurred.
What should i do？

[image]
[image]

### Include context on what you are trying to achieve

#### Context details

### Include details of what you already did to find answers

jonyMarino, 2021-10-15:
Hi @Hijiongfeng! I've never done that, but I'm going to try to help you. What I see is that it doesn't find the libraries that are compiled when working on Linux. Can you compile AirSim inside a WSL environment and copy the .a files missing for the packaging?

Hijiongfeng, 2021-10-15:
> Hi @Hijiongfeng! I've never done that, but I'm going to try to help you. What I see is that it doesn't find the libraries that are compiled when working on Linux. Can you compile AirSim inside a WSL environment and copy the .a files missing for the packaging?

Hi, thank you for your help. I previously compiled AirSim successfully on Ubuntu, I copied the missing .a file to the corresponding directory, and then packaged it successfully. I want to know if I can not use this copy method, it is too cumbersome, can I directly use a cross-compilation method to operate. If so, what should I do?

jonyMarino, 2021-10-15:
You can use WSL2 inside windows as an alternative.

closed 2022-08-01 by jonyMarino