microsoft/AirSim #4081: AirLib.lib(RpcLibClientBase.obj) : error LNK2019 when trying to install AirSim in Windows

martin0004, 2021-10-13:
## Bug report 

- AirSim Version/#commit: latest
- UE/Unity version: N/A
- autopilot version: N/A
- OS Version: Windows 10

### What's the issue you encountered?

I am trying to build AirSim in Windows 10 by following the procedure described here: https://microsoft.github.io/AirSim/build_windows/.

I get the following error message. Any idea on how to solve this?

[image]

### Settings

### How can the issue be reproduced?

1. My coworkers get the same error messages.

### Include full error message in text form

12>AirLib.lib(RpcLibClientBase.obj) : error LNK2019: unresolved external symbol "public: __cdecl rpc::client::clien
       t(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,unsigned short
       )" (??0client@rpc@@QEAA@AEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@G@Z) referenced in func
       tion "public: __cdecl msr::airlib::RpcLibClientBase::impl::impl(class std::basic_string<char,struct std::char_tr
       aits<char>,class std::allocator<char> > const &,unsigned short,float)" (??0impl@RpcLibClientBase@airlib@msr@@QEA
       A@AEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@GM@Z) [F:\Users\martin\airsim\AirSim\HelloCar
       \HelloCar.vcxproj]
    12>AirLib.lib(RpcLibClientBase.obj) : error LNK2019: unresolved external symbol "public: __cdecl rpc::client::~clie
       nt(void)" (??1client@rpc@@QEAA@XZ) referenced in function "int `public: __cdecl msr::airlib::RpcLibClientBase::i
       mpl::impl(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,unsign
       ed short,float)'::`1'::dtor$0" (?dtor$0@?0???0impl@RpcLibClientBase@airlib@msr@@QEAA@AEBV?$basic_string@DU?$char
       _traits@D@std@@V?$allocator@D@2@@std@@GM@Z@4HA) [F:\Users\martin\airsim\AirSim\HelloCar\HelloCar.vcxproj]
    12>AirLib.lib(RpcLibClientBase.obj) : error LNK2019: unresolved external symbol "public: class nonstd::optional_lit
       e::optional<__int64> __cdecl rpc::client::get_timeout(void)const " (?get_timeout@client@rpc@@QEBA?AV?$optional@_
       J@optional_lite@nonstd@@XZ) referenced in function "public: class clmdep_msgpack::v1::object_handle __cdecl rpc:
       :client::call<>(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &)
       " (??$call@$$V@client@rpc@@QEAA?AVobject_handle@v1@clmdep_msgpack@@AEBV?$basic_string@DU?$char_traits@D@std@@V?$
       allocator@D@2@@std@@@Z) [F:\Users\martin\airsim\AirSim\HelloCar\HelloCar.vcxproj]
(...)

**What's better than filing an issue? Filing a pull request :).**

jonyMarino, 2021-10-15:
The check that runs the `build.cmd` in our pipelines is ok...
https://github.com/microsoft/AirSim/runs/3900944502
Maybe you need to run `clear.cmd` before?

zimmy87, 2022-01-14:
Hey @martin0004 are you still experiencing this issue?

martin0004, 2022-01-18:
Hi @zimmy87 . My apologies for my late response. I switched to a Linux machine running Ubuntu 20.04. AirSim is working fine now in Linux. I'll close the issue.

closed 2022-01-18 by martin0004