microsoft/AirSim #4087: Image API seems to block between WSL and Windows

ab3nd, 2021-10-15:
## Bug report 

- AirSim Version/#commit: v1.6.0-windows
- UE/Unity version:  4.26
- autopilot version: N/A
- OS Version: Ubuntu 18.04.5 LTS (In WSL) and Windows 10 Pro

### What's the issue you encountered?

The Unreal Editor is being run in Windows, but the code to interact with it is being run from WSL on the same computer. The simGetImages and simGetImage calls from a python script hang indefinitely without returning. When operating the AirSim API from an interactive python session (see below), some of the other camera-related API calls do work (can get distortion parameters and camera info for the external camera). Moving the camera also works, or at least returns after being called. 

'''
(base) abend@hulk:/mnt/e/SSCI_Environments/Scripts/data_collection$ python3
Python 3.8.5 (default, Sep  4 2020, 07:30:14)
[GCC 7.3.0] :: Anaconda, Inc. on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import airsim
>>> client = airsim.VehicleClient() #NOT MultirotorClient
>>> client.confirmConnection()
Connected!
Client Ver:1 (Min Req: 1), Server Ver:1 (Min Req: 1)

>>> airsim.__version__
'1.6.0'
>>> cam_info = client.simGetCameraInfo("FixedCamera1", external=True)
>>> print(cam_info)
<CameraInfo> {   'fov': 90.0,
    'pose': <Pose> {   'orientation': <Quaternionr> {   'w_val': 0.5457270741462708,
    'x_val': -0.0008777615730650723,
    'y_val': -0.8379622101783752,
    'z_val': -0.0005715602892450988},
    'position': <Vector3r> {   'x_val': -83.28520202636719,
    'y_val': 53.362449645996094,
    'z_val': 2.808880567550659}},
    'proj_mat': <ProjectionMatrix> {   'matrix': [   [   -0.0,
                      3.866713047027588,
                      -0.0,
                      0.0],
                  [   -0.0,
                      0.0,
                      -6.874156475067139,
                      0.0],
                  [   -0.0,
                      0.0,
                      -0.0,
                      10.0],
                  [   -1.0,
                      0.0,
                      -0.0,
                      0.0]]}}
>>> requests = [airsim.ImageRequest("FixedCamera1", airsim.ImageType.Scene)]
>>> responses = client.simGetImages(requests, external=True)
^CTraceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/abend/anaconda3/lib/python3.8/site-packages/airsim/client.py", line 266, in simGetImages
    responses_raw = self.client.call('simGetImages', requests, vehicle_name, external)
  File "/home/abend/anaconda3/lib/python3.8/site-packages/msgpackrpc/session.py", line 41, in call
    return self.send_request(method, args).get()
  File "/home/abend/anaconda3/lib/python3.8/site-packages/msgpackrpc/future.py", line 25, in get
    self.join()
  File "/home/abend/anaconda3/lib/python3.8/site-packages/msgpackrpc/future.py", line 22, in join
    self._loop.start()
  File "/home/abend/anaconda3/lib/python3.8/site-packages/msgpackrpc/loop.py", line 22, in start
    self._ioloop.start()
  File "/home/abend/anaconda3/lib/python3.8/site-packages/tornado/ioloop.py", line 863, in start
    event_pairs = self._impl.poll(poll_timeout)
KeyboardInterrupt
>>> del client
>>> client = airsim.VehicleClient() #NOT MultirotorClient
>>> png_image = client.simGetImage("FixedCamera1", airsim.ImageType.Scene)
[... 306 lines cut ...]
          "MotionBlurAmount": 0
        },
        {
          "ImageType": 1,
          "Width": 1280,
          "Height": 720,
          "FOV_Degrees": 29,
  
          "MotionBlurAmount": 0
        }
      ]
    }
  }
}

```
### How can the issue be reproduced?

1. Start AirSim with the above settings.json
2. Start a python3 interactive shell in WSL
3. Attempt calls to simGetImage or simGetImages as in the copy/pasted shell above

### Include full error message in text form

No error message, function just doesn't return.

jonyMarino, 2021-10-19:
Hi @ab3nd! Can you try with WSL2 instead of WSL? You would need to pass the host IP to the client.

ab3nd, 2022-01-04:
This may have been related to Symantec Endpoint Protection, which will block WSL2 but not WSL1. Might also have been version mismatches in my scripts vs. the installed version of AirSim. I didn't fix it, just moved on to other things.

closed 2022-01-04 by ab3nd