microsoft/AirSim #4092: Spawning multiple "ComputerVision" pawns results in actor name conflict.

Octavian-Zhang, 2021-10-19:
## Bug report 
- AirSim Version/#commit: [f1ccd2d109073e2b16506a5ed31240ffad92d815](https://github.com/microsoft/AirSim/commit/f1ccd2d109073e2b16506a5ed31240ffad92d815)
- UE/Unity version: UE4.27
- autopilot version: N/A
- OS Version: Windows 10 21H1

### What's the issue you encountered?
I'm trying to spawn multiple "ComputerVision" pawns using the following settings.json content:
`{
  "SettingsVersion": 1.2,
  "SimMode": "ComputerVision",
  "Vehicles": {
    "Cam1": {
      "VehicleType": "ComputerVision",
      "X": 26,
      "Y": -1,
      "Z": -10
    },
    "Cam2": {
      "VehicleType": "ComputerVision",
      "X": 26,
      "Y": -2,
      "Z": -20
    }
  }
}`
The compiled Unreal executable fails to start with the following error prompt:
[image]

### Settings
The settings.json is as this short as possible:
`{
  "SettingsVersion": 1.2,
  "SimMode": "ComputerVision",
  "Vehicles": {
    "Cam1": {
      "VehicleType": "ComputerVision",
      "X": 26,
      "Y": -1,
      "Z": -10
    },
    "Cam2": {
      "VehicleType": "ComputerVision",
      "X": 26,
      "Y": -2,
      "Z": -20
    }
  }
}`
If only one instead of two "ComputerVision" pawn is to be spawned at Unreal start, the error above disappears. Try to delete "Cam2" content from the settings above to see the effect.

### How can the issue be reproduced?
1. Place the settings.json file above under Documents\AirSim.
2. Down and run any precompiled [Unreal binary executable](https://github.com/Microsoft/AirSim/releases).

### Include full error message in text form
An actor name 'camera_front_center' already exists in level 'Level/Game/Maps/LandscapeMap.LandscapeMap:PersistentLevel'.

jonyMarino, 2021-10-19:
I confirm the bug. I tried version 1.2.2, and it didn't happen. Is it suitable for you?

Octavian-Zhang, 2021-10-20:
> I confirm the bug. I tried version 1.2.2, and it didn't happen. Is it suitable for you?

Sure, @jonyMarino, thanks for providing the workaround. It solves my problem. Cheers!

PR #4109 (2021-12-08): Fix actor name conflict for multiple ComputerVision vehicles

Fixes: #4092    

## About

Fixes the name conflict by including the vehicle name

## How Has This Been Tested?

With the below settings from #4092 
```json
{
    "SettingsVersion": 1.2,
    "SimMode": "ComputerVision",
    "Vehicles": {
        "Cam1": {
            "VehicleType": "ComputerVision",
            "X": 26,
            "Y": -1,
            "Z": -10
        },
        "Cam2": {
            "VehicleType": "ComputerVision",
            "X": 26,
            "Y": -2,
            "Z": -20
        }
    }
}
```

## Screenshots [... cut]
closed 2021-12-08 by zimmy87