microsoft/AirSim #4138: GPS prearm errors when arming with MavProxy

khnunn, 2021-10-30:
## Bug report 

- AirSim Version/#commit: #4102
- UE/Unity version: 4.27.1
- autopilot version: ArduCopter 4.1
- OS Version: Windows 10

### What's the issue you encountered?

Hi, I’m running ArduCopter 4.1 on a black cube through MaxProxy and AirSim. When start AirSim I get the following response in the MavLink console:

```
pre-arm good
AP: EKF3 IMU0 origin set
AP: EKF3 IMU1 origin set
Got COMMAND_ACK: DO_SET_MODE: ACCEPTED
Mode GUIDED
pre-arm fail
AP: PreArm: GPS 1: not healthy
Got COMMAND_ACK: COMPONENT_ARM_DISARM: FAILED
AP: PreArm: GPS 1: not healthy
AP: EKF3 IMU0 is using GPS
AP: EKF3 IMU1 is using GPS
Flight battery 100 percent
AP: PreArm: GPS 1: not healthy
```

Mission planner parameters were taken from Tools/autotest/default_params/copter.param.

### Settings

```
{
  "SettingsVersion": 1.2,
  "LogMessagesVisible": true,
  "SimMode": "Multirotor",
  "OriginGeopoint": {
    "Latitude": -35.363261,
    "Longitude": 149.165230,
    "Altitude": 583
  },
  "Vehicles": {
    "Copter": {
      "VehicleType": "ArduCopter",
      "UseSerial": false,
      "LocalHostIp": "127.0.0.1",
      "UdpIp": "127.0.0.1",
      "UdpPort": 9003,
      "ControlPort": 9002
    }
  }
}
```

### How can the issue be reproduced?

1. Upload ArduCopter 4.1 using Mission Planner 1.3.75 to a PixHawk Black on a Windows 10 machine. Connect to the device under Simulation in Mission Planner.
2. Run the following commands in CygWin:
```
cd ArduCopter
../Tools/autotest/sim_vehicle.py -v ArduCopter -f airsim-copter --console --map
```
3. Start AirSim.

### Include full error message in text form

**What's better than filing an issue? Filing a pull request :).**

rajat2004, 2021-10-30:
Hi, can you try arming in Stabilize mode instead of Guided or Loiter (basically any GPS mode). Not sure why but this happening with some other simulators also. If it still doesn't work, maybe try disabling the GPS arming check also

khnunn, 2021-10-30:
> 
> 
> Hi, can you try arming in Stabilize mode instead of Guided or Loiter (basically any GPS mode). Not sure why but this happening with some other simulators also. If it still doesn't work, maybe try disabling the GPS arming check also

Hi Rajat, ARMING_CHECK is set to 0 for no checks. I get the following console output when I set mode to stabilize and arm throttle:
```
Got COMMAND_ACK: DO_SET_MODE: ACCEPTED
Mode STABILIZE
pre-arm good
Got COMMAND_ACK: COMPONENT_ARM_DISARM: FAILED
AP: PreArm: Throttle too high
```

rajat2004, 2021-11-01:
Sorry for the late reply, try setting the throttle to 1000 as well

khnunn, 2021-11-05:
I did a clean pull of AirSim, ardupilot, and created a new Unreal project and the issue has disappeared. Sorry for not having a specific solution to the problem.

closed 2021-11-05 by khnunn