microsoft/AirSim #4142: Receiving compressed image when settings are on uncompressed

imanf94, 2021-11-03:
## Bug report 

- AirSim Version/#commit: 1.2
- UE/Unity version: Using stand-alone City Environment
- autopilot version: N/A
- OS Version: Windows 10

### What's the issue you encountered?

I want to get uncompressed images from image API but it returns compressed images. I tried to change the args and the settings.json to get uncompressed images but it couldn't help. I expect a 1280*720 image but the output is a 255*160ish image.
### Settings

{
"SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
"SettingsVersion": 1.2,
"SimMode": "Car",
"Vehicles": {
    "PhysXCar": {
      "VehicleType": "PhysXCar",
      "Cameras": {   
        "CAM0": {
          "X": 1, "Y": 0, "Z": -1.5,
		  "Pitch": -10, "Roll": 0, "Yaw": 0,
		  "ImageType": 1,
		  "PixelsAsFloat": false,
		  "Compress": false
        }
      }
    }
  },
"CameraDefaults": {
"CaptureSettings": [
{
"ImageType": 0,
"Width": 1280,
"Height": 720,
"FOV_Degrees": 120
}
]
}
}
### How can the issue be reproduced?

The name of the camera is CAM0 and here is my code:

responses = client.simGetImages([airsim.ImageRequest("CAM0", airsim.ImageType.Scene, False, False)])

    #responses = client.simGetImages([airsim.ImageRequest("0", airsim.ImageType.DepthPlanner, True)])
    for response in responses:
        img1d = np.fromstring(response.image_data_uint8, dtype=np.uint8) 
        
        # reshape array to 4 channel image array H X W X 4
        img_rgb = img1d.reshape(response.height, response.width, 3)
        plt.imshow(img_rgb)
        plt.show()

rajat2004, 2021-11-04:
I think the internal implementation uses the `CameraDefault` settings only if the entire setting for the camera is missing. Since there are some settings present for `CAM0`, it's using those alongwith with default values of height, width, etc.  Try specifying the H,W as well in the `CAM0` setting.
Changing this behaviour inside the code might be a bit messy, and IMO it's easier to figure out the exact settings from one place rather than piecing them together from multiple, but upto the maintainers. In any case, the documentation should definitely be updated to clarify the intention

imanf94, 2021-11-04:
Thanks, @rajat2004 for your reply. This couldn't help am I'm still getting compressed images.
Here is my new setting:
```

{
"SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
"SettingsVersion": 1.2,
"SimMode": "Car",
"Vehicles": {
    "PhysXCar": {
      "VehicleType": "PhysXCar",
      "Cameras": {   
        "CAM0": {
          "X": 1, "Y": 0, "Z": -1.5,
		  "Pitch": -10, "Roll": 0, "Yaw": 0,
		  "ImageType": 0,
		  "PixelsAsFloat": false,
		  "Compress": false,
		  "Width": 1280,
	      "Height": 720
        }
      }
    }
  },
"CameraDefaults": {
"CaptureSettings": [
{
"ImageType": 0,
"Width": 1280,
"Height": 720,
"FOV_Degrees": 120
}
]
}
}

```
And here is my code:

```
    responses = client.simGetImages([airsim.ImageRequest("CAM0", airsim.ImageType.Scene, False, False)])

    for response in responses:
        img1d = np.fromstring(response.image_data_uint8, dtype=np.uint8) 
        
        # reshape array to 4 channel image array H X W X 4
        img_rgb = img1d.reshape(response.height, response.width, 3)
```

I need to mention that if I change the Camera name to 1 (default front camera) as below I get uncompressed images. However, the pinch and the position is not applied to these images:
` responses = client.simGetImages([airsim.ImageRequest("CAM0", airsim.ImageType.Scene, False, False)])`

Any idea how I can get uncompressed images from a camera with customized position and orientation?

rajat2004, 2021-11-05:
The `CAM0` setting should also be in the same format as `CameraDefaults`, something like -
```json
"Cameras": {   
  "CAM0": {
    "CaptureSettings": [
        {
        ...
        }
     ]
  }
}
```

imanf94, 2021-11-07:
Thanks, @rajat2004! That fixed the issue.

closed 2021-11-07 by imanf94