microsoft/AirSim #4144: Thrust Loss Error Prevents Arming with SITL

khnunn, 2021-11-05:
## Question
### What's your question?
I had no problems running AirSim, until I connected a transmitter (Taranis X9D+ 2019) to my PixHawk 2.1 and performed radio calibrations. After I loaded sim_vehicle.py with these parameters, I started getting this error message in the MavProxy console: "AP: Potential Thrust Loss (6)". The rotors do not spin up in the simulation, and the drone does not respond to takeoff commands or serial RC control. I have tried setting the parameter "FLIGHT_OPTIONS" to 3 for no thrust loss checks, but the drone still does not respond. Does anyone know which parameter setting I need to fix to remove this message? 

#### Context details

OS: Windows 10
AirSim v1.6.0
ArduCopter v4.1.0
Mission Planner 1.3.75
UE4 v4.27.1

These are my AirSim settings and ArduCopter parameters (rename from .txt to .json and .parm respectively):
[settings.json](https://github.com/microsoft/AirSim/files/7486127/settings.txt)
[parameters.parm](https://github.com/microsoft/AirSim/files/7486130/parameters.txt)

rajat2004, 2021-11-05:
I don't think AP-AirSim integration works with Hexacopter, or at least I haven't tried with it. Was it working properly without the transmitter, i.e. arming and flying through Mavproxy? If not, then that needs to be fixed first
With transmitter, it might be the case that AirSim simply doesn't work with it. Have you tried using the RC with say SimpleFlight or PX4 and checked that it works? I haven't played with RCs so not much idea about this. If it does work, then the next steps would be to see the packets being sent to and fro, using Wireshark

khnunn, 2021-11-05:
> I don't think AP-AirSim integration works with Hexacopter, or at least I haven't tried with it. Was it working properly without the transmitter, i.e. arming and flying through Mavproxy? If not, then that needs to be fixed first
Indeed, the simulation was working without the transmitter problems after following Chris Lovett's [guide ](https://github.com/microsoft/AirSim/wiki/hexacopter), with few modifications.
> With transmitter, it might be the case that AirSim simply doesn't work with it.
The problem persists whether the RC is connected or not. Previously I was able to arm and fly the drone with the transmitter (Taranis X9D+ 2019) connected through serial port, but that no longer works. Nor does the drone arm when I only use MavProxy commands, without any RC. I have tried restoring the original "copter.parm" parameters I used for my original simulation, but I run into the same error. 

In Mission Planner, adjusting attitude and throttle does not seem to affect the PWM outputs to my motor servos. I suspect this might be related to the cause of this error, but I don't know how it relates to the AirSim model.
[image]

khnunn, 2021-11-06:
I have resolved the issue. AirSim was not rebuilding EEPROM whenever I loaded a new set of parameters from a file. Adding the -w command to "../Tools/autotest/sim_vehicle.py -v ArduCopter -f airsim-copter --console --map -w" reset EEPROM and allowed me to define new parameters.

closed 2021-11-06 by khnunn