microsoft/AirSim #4148: Install airsim on cpu only computer

Flautdn, 2021-11-06:
Can I work with airsim simulations on ubuntu 18 with cpu only?

When it comes to aws ec2 cpu only instance in free tier section as t2.micro instance, I am able to install
Airsim binary but when executing the AirsimNH (city environment) I am getting 
unable cretae opengl context error. 

Ubuntu 18, AirSim 1.6, Python 3, Unreal 4

Thank you.

jonyMarino, 2021-11-22:
@Flautdn Even if it would be possible, the result would be excessively slow.

jonyMarino, 2021-11-22:
There is an azure environment ready for use with AirSim:
https://microsoft.github.io/AirSim/azure/

rhalaly, 2022-01-16:
@jonyMarino I'm facing the exact same issue. I'm using Azure but my algorithms heavily depend on the CPU. The recommendation in https://microsoft.github.io/AirSim/azure/ is a GPU machine (NV-machines), but I prefer to buy more CPUs without "unnecessary" GPUs.

For example, NV24 has 24 vCPUs but brings 4x unnecessary GPUs. But F72s v2 costs around the same price, but it has 72 vCPUs!

About the slowness, I run AirSim in "synced" mode (with `simContinueForTime` - calculate and run the simulation), so I don't care that simulation will be slow.

So, is there an option to run AirSim without any GPUs?

xxEoD2242, 2022-01-16:
@rhalaly Unreal Engine, not AirSim, uses the GPU and that is required regardless of how you run the system. As AirSim is a plugin for UE, it would not be possible to use this without a GPU available.

If you are savy with networking, I would recommend running UE on a low-cost GPU instance and then network in your algorithms on a CPU-only instance through the APIs. Since Msgpack is inherently a comms system, you can network together instances to make that happen. There will be a delay but 2 instances on the same timezone in Azure should have very minimal latency.

rhalaly, 2022-01-16:
> Unreal Engine, not AirSim, uses the GPU and that is required regardless of how you run the system. As AirSim is a plugin for UE, it would not be possible to use this without a GPU available.
Of course :) 

Actually, it is an interesting idea! Since I run the simulation in "sync mode" (send command then simulate) I don't think I will affect by latency. 
But the bottleneck for parallelism is still the GPU machines - how much simulation you can run in parallel. With a low-cost GPU, the answer will probably be 1...

My goal is to use Azure Batch for running multiple tasks with different hyperparameters.