microsoft/AirSim #4154: `simDestroyObject('CityEnviron_NPCVehicleManager')` Causes the simulator to crash

nooreldeensalah, 2021-11-09:
## Bug report 

- AirSim Version/#commit: 1.60
- UE/Unity version: N/A
- autopilot version: N/A
- OS Version: Windows 10, Version 21H1, OS Build (19043.1288).

### What's the issue you encountered?

I'm using a pre-compiled binary of the simulator, specifically the `CityEnviron` binary, I wanted to remove all moving vehicles and pedestrians from the environment, because I'm interested in capturing images without the vehicles on the road, nor the pedestrians.

So my attempt was to call `simDestroyObject` on both the `CityEnv_PedManager_26` and `CityEnviron_NPCVehicleManager` objects.

Calling `simDestroyObject('CityEnv_PedManager_26)` works as intended, it removes the pedestrians from the environment, but calling `client.simDestroyObject('CityEnviron_NPCVehicleManager')` causes the simulator to crash as shown in the next image:

[image]

### Settings

```json
{
  "SettingsVersion": 1.2,
  "SimMode": "Car"
}
```

### How can the issue be reproduced?

1. Open the AirSim pre-compiled `CityEnviron` binary.
2. Execute the following lines
```python
import airsim
client = airsim.VehicleClient()
client.simDestroyObject('CityEnv_PedManager_26')
client.simDestroyObject('CityEnviron_NPCVehicleManager')
```
### Include full error message in text form
`Fatal Error` with the simulator crashing on `client.simDestroyObject('CityEnviron_NPCVehicleManager')` method call.

jonyMarino, 2021-11-10:
@nooreldeensalah v1.5 is suitable for you?

nooreldeensalah, 2021-11-11:
@jonyMarino 
I apologize for the late reply,
I've downloaded v1.5, and I think it's suitable for my purposes, because it has no pedestrians nor vehicles.

But with that being said, feel free to keep or close the issue depending on if this bug can be fixed or not.

nooreldeensalah, 2021-11-11:
@jonyMarino 
One last thing, I had a script that I was using to collect images https://github.com/ElGarash/AirSim-Localization/blob/main/script.py

The idea is, it worked well with v1.6.0, but seems like I need to modify it for v1.5.0, where can I find the API reference docs for v1.5.0?

nooreldeensalah, 2021-11-11:
@jonyMarino 
One last thing, I had a [script ](https://github.com/ElGarash/AirSim-Localization/blob/main/script.py)that I was using to collect images, the script requires an external camera (what the script briefly does is; get the pose of the ground vehicle as it's moving, adjust the pose of the external aerial camera accordingly, then it captures images from the ground camera attached to the vehicle, and the aerial external camera).

The script worked well for airsim v1.6.0, but it's now causing the simulator to crash, I've degraded my airsim pip package to 1.5.0 and after some investigation, turned out that external cameras aren't supported in v1.5.0 as in #3320 (which was added after the v1.5.0 release).

Any tips on how to handle this case? (i.e. taking aerial images for the vehicle)

zimmy87, 2021-11-18:
Hi @nooreldeensalah, thank you for reporting this issue. This is happening because car and bicycle paths in the city environment store a reference to the CityEnviron_NPCVehicleManager instance and call methods on it during their Tick() method, so destroying the CityEnviron_NPCVehicleManager instance immediately causes a nullptr exception. To remove the CityEnviron_NPCVehicleManager, you'll need to first remove all intersection logic, bicycle paths, and car paths. The following script works for me locally:

```
import airsim
client = airsim.VehicleClient()
client.simDestroyObject("CityEnv_PedManager_26")

intersections = client.simListSceneObjects("BaseIntersectionLogic_BP.*")
for intersection in intersections:
    client.simDestroyObject(intersection)

bike_paths = client.simListSceneObjects("BaseBaseBicyclePathBP.*")
for bike_path in bike_paths:
    client.simDestroyObject(bike_path)

car_paths = client.simListSceneObjects("BaseCarPath_BP.*")
for car_path in car_paths:
    client.simDestroyObject(car_path)
    
client.simDestroyObject('CityEnviron_NPCVehicleManager')
```

closed 2021-11-18 by jonyMarino