microsoft/AirSim #4158: Linux development workflow

talha-opteran, 2021-11-12:
## Question
### What's your question?

I'm confused about what the development workflow looks like for AirSim. Some clarification on how others are approaching this would be nice.

### Include context on what you are trying to achieve

For context, I'm looking to extend the Unreal plugin which requires modifying code in:
```
Blocks\Plugins\AirSim\Source
Blocks\Plugins\AirSim\Source\Airlib
```

#### Context details

OS: Ubuntu 20.04
AirSim: git master
Unreal: git master (4.27.1)
IDE: VSCode

### Include details of what you already did to find answers
From my understanding of the docs, there are 2 approaches:

1. As described in the docs, [[1]](https://microsoft.github.io/AirSim/build_linux/)[[2]](https://microsoft.github.io/AirSim/dev_workflow/#steps), this workflow is as follows:
```
git pull                          
build.sh                       
cd Unreal\Environments\Blocks         
update_from_git.sh
**Modify code in the above directories**
update_to_git.sh
build.sh
```
However, as mentioned in #2975, `update_to_git.sh` deletes the `Airlib\src` directory in the AirSim repo. This also causes the subsequent `build.sh` to fail. 

2. In this approach, [[3]](https://microsoft.github.io/AirSim/unreal_blocks/#changing-code-and-rebuilding_1), it recommends developing in the root `Airlib` as well as `Unreal\Plugins` directories directly followed by running `build.sh` to copy the changes to the Unreal project.   

This works well but is tedious as it requires manual setup to work with IDE features like intellisense.

So what is the official approach and what are others using? Thanks

rajat2004, 2021-11-21:
I personally use the 2nd approach of working on the root `AirLib` and `Unreal/Plugins` dir. It has a benefit for quicker testing of AirLib changes for compilation issues first, and then going for UE4. Haven't found Intellisense or other IDE features to be very good tbh since they don't work with UE4 headers and code

jonyMarino, 2021-11-22:
@talha-opteran on windows, I prefer to use the first approach. That is because I use visual studio with the debugger in Blocks project. Find no problems for now in using update_to_git.
In Linux, I think the second approach is more straightforward.

talha-opteran, 2021-11-23:
Thanks for the input. 

@jonyMarino I think the reason you don't have problems in windows with `update_to_git.bat` is because it is using `robocopy`. Not sure how that works but Linux uses `rsync` with the `--delete` option which causes the `Airlib\src` folder to be deleted.