microsoft/AirSim #4160: DQN_car

DAndrik21, 2021-11-14:
## Question
### What's your question?
Im running dqn_car.py.Training started at 200k timesteps as it should and for 10k steps it was working(i had good reward) but from then until now (50k steps ) the car just turns right (steering either 0.25 or 0.5) and resets immediately .

### Include context on what you are trying to achieve
Has anyone run this ?is it normal to do this for so long?

#### Context details
Windows 11 ,AirSimNH, Python 3.9.0

### Include details of what you already did to find answers

DAndrik21, 2021-12-02:
Can anyone confirm dqn_car.py work after 210k timesteps?

Zhenlin-Xu, 2022-01-07:
Did you check the cumulative reward of every episode on Tensorboard?  I start a training on dqn_car.py today. I will go back to this issue if I find something interesting.