microsoft/AirSim #4171: Camera positions is not synchronized across multiple cameras

suresh-guttikonda, 2021-11-19:
## Bug report 

- AirSim Version/#commit: master branch
- UE/Unity version: UnrealEngine_4.25
- autopilot version:
- OS Version: Ubuntu 18.04.6 LTS

### What's the issue you encountered?

Its very annoying and there are several bugs/issues still open for couple of years [#2784](https://github.com/microsoft/AirSim/issues/2784) still there is no permanent solution and AirSim team didn't provide update on bug till now. I tried other simulator like Carla which is also based on UE, where it is able to sync correctly across multiple images without any issue !!!

I am opening this new issue/bug to bring this to attention for developers.

### Settings

```
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Car",
  "ViewMode": "SpringArmChase",
  "ClockSpeed": 1.0,
  "Vehicles": {
    "drone_1": {
      "VehicleType": "PhysXCar",
      "DefaultVehicleState": "Armed",
      "EnableCollisionPassthrogh": false,
      "EnableCollisions": true,
      "AllowAPIAlways": true,
      "Sensors": {
        "Gps": {
          "SensorType": 3,
          "Enabled" : true
        },
        "Barometer": {
          "SensorType": 1,
          "Enabled" : true
        },
        "Magnetometer": {
          "SensorType": 4,
          "Enabled" : true
        },
        "Imu" : {
          "SensorType": 2,
          "Enabled": true
        },        
        "LidarCustom": {
          "SensorType": 6,
          "Enabled": true,
          "NumberOfChannels": 16,
          "PointsPerSecond": 10000,
          "X": 0,
          "Y": 0,
          "Z": -1.5,
          "DrawDebugPoints": true,
          "DataFrame": "SensorLocalFrame"
        }
      },
      "Cameras": {
        "front_center_custom": {
          "CaptureSettings": [
            {
              "PublishToRos": 1,
              "ImageType": 0,
              "Width": 640,
              "Height": 480,
              "FOV_Degrees": 27,
              "DepthOfFieldFstop": 2.8,
              "DepthOfFieldFocalDistance": 200.0, 
              "DepthOfFieldFocalRegion": 200.0,
              "TargetGamma": 1.5
            }
          ],
          "X": 1.75, "Y": 0, "Z": -1.25,
          "Pitch": 0, "Roll": 0, "Yaw": 0
        },
        "front_left_custom": {
          "CaptureSettings": [
            {
              "PublishToRos": 1,
              "ImageType": 0,
              "Width": 672,
              "Height": 376,
              "FOV_Degrees": 90,
              "TargetGamma": 1.5
            }
          ],
          "X": 1.75, "Y": -0.5, "Z": -1.25,
          "Pitch": 0.0, "Roll": 0.0, "Yaw": 0.0
        },
        "front_right_custom": {
          "CaptureSettings": [
            {
              "PublishToRos": 1,
              "ImageType": 0,
              "Width": 672,
              "Height": 376,
              "FOV_Degrees": 90,
              "TargetGamma": 1.5
            }
          ],
          "X": 1.75, "Y": 0.5, "Z": -1.25,
          "Pitch": 0.0, "Roll": 0.0, "Yaw": 0.0
        }
      },
      "X": 0, "Y": 0, "Z": 0,
      "Pitch": 0, "Roll": 0, "Yaw": 0
    }
  },
  "SubWindows": [
    {"WindowID": 0, "ImageType": 0, "CameraName": "front_left_custom", "Visible": false},
    {"WindowID": 1, "ImageType": 0, "CameraName": "front_center_custom", "Visible": false},
    {"WindowID": 2, "ImageType": 0, "CameraName": "front_right_custom", "Visible": false}
  ]
}

```

### How can the issue be reproduced?

1.
```
    // update left+right images from airsim_client_ every nth sec
    void image_response_cb(const ros::TimerEvent& event) {
      std::vector<ImageRequest> request = {
        ImageRequest("front_left_custom", ImageType::Scene),
        ImageRequest("front_center_custom", ImageType::Scene),
        ImageRequest("front_right_custom", ImageType::Scene)
      };
      const std::vector<ImageResponse>& response = airsim_client_.simGetImages(request);
      std::cout << "Left Image  [" << left_cnt_ << "]: " <<
        convert_airsim_timestamp(response[0].time_stamp).count() << std::endl <<
        response[0].camera_position << std::endl;
      left_cnt_ += 1;
      std::cout << "Center Image[" << center_cnt_ << "]: " <<
        convert_airsim_timestamp(response[1].time_stamp).count() << std::endl <<
        response[1].camera_position << std::endl;
      center_cnt_ += 1;
      std::cout << "Right Image [" << right_cnt_ << "]: " <<
        convert_airsim_timestamp(response[2].time_stamp).count() << std::endl <<
        response[2].camera_position << std::endl;
      right_cnt_ += 1;
      std::cout << "------------------" << std::endl;
    }
```

2. When i move the car around the UE simulator using keyboard the camera positions are not in sync. ideally the x & z position should be in sync for stereo setup. I also tried the pauseSim solution which suggested in other posts but didn't work.

### Include full error message in text form

```
Left Image  [58]: 1637329684254
    1.75
-0.49992
-1.01494
Center Image[58]: 1637329684258
       1.75
7.99627e-05
   -1.01491
Right Image [58]: 1637329684262
    1.75
 0.50008
-1.01488
------------------
Left Image  [59]: 1637329684703
 1.75061
-0.49992
-1.01218
Center Image[59]: 1637329684706
    1.75061
7.98117e-05
   -1.01215
Right Image [59]: 1637329684708
 1.75061
 0.50008
-1.01212
------------------
Left Image  [60]: 1637329685081
  1.69931
-0.497714
 -1.00851
Center Image[60]: 1637329685084
   1.69865
0.00228599
  -1.00848
Right Image [60]: 1637329685086
 1.69799
0.502286
-1.00845
------------------
Left Image  [61]: 1637329685523
   1.2688
-0.498193
 -1.00507
Center Image[61]: 1637329685525
   1.26811
0.00180677
  -1.00504
Right Image [61]: 1637329685527
 1.26742
0.501806
-1.00501
------------------
Left Image  [62]: 1637329685932
 0.472647
-0.499356
 -1.00516
Center Image[62]: 1637329685936
   0.471978
0.000643664
   -1.00513
Right Image [62]: 1637329685939
0.471309
0.500643
 -1.0051
```
**What's better than filing an issue? Filing a pull request :).**

NickPerezCarletonUniversity, 2021-11-19:
you have simPause commented out

suresh-guttikonda, 2021-11-19:
@NickPerezCarletonUniversity I know, as I mentioned if i use simPause with ros::AsyncSpinner for > 10Hz fps the UE4 is not accepting keyboard inputs to control default car. Hence i commented it out.

suresh-guttikonda, 2021-11-19:
Alternative test case, when the car is moving in x direction i.e forward @ 20 m/s speed in straight path, run the following code and see the results.
Note: Use less CPU in Background is unchecked.

```
#!/usr/bin/env python3
# -*- coding: utf-8 -*-

import airsim #pip install airsim

airsim_client = airsim.CarClient()

airsim_client.simPause(True);
response = airsim_client.simGetImages([
    airsim.ImageRequest("front_left_custom", airsim.ImageType.Scene),
    airsim.ImageRequest("front_center_custom", airsim.ImageType.Scene),
    airsim.ImageRequest("front_right_custom", airsim.ImageType.Scene)
])
airsim_client.simPause(False);

print(f"Left Image   : {response[0].time_stamp} {response[0].camera_position}");
print(f"Center Image : {response[1].time_stamp} {response[1].camera_position}");
print(f"Right Image  : {response[2].time_stamp} {response[2].camera_position}");
```

output:
```
Left Image   : 1637356737920371000 <Vector3r> {   'x_val': -23.21270179748535,
    'y_val': -0.25999313592910767,
    'z_val': -1.007110834121704}
Center Image : 1637356737935430000 <Vector3r> {   'x_val': -23.21417999267578,
    'y_val': 0.24000467360019684,
    'z_val': -1.0070784091949463}
Right Image  : 1637356737944388000 <Vector3r> {   'x_val': -23.215654373168945,
    'y_val': 0.7400025129318237,
    'z_val': -1.007046103477478}
```

[image]