microsoft/AirSim #4177: Move to Target Velocity

Gastastrophe, 2021-11-24:
## Question
### What's your question?

Hello all. I am trying to implement Proportional Navigation using a Line of Sight vector, and part of that involves applying a calculated acceleration to the drone. I was planning to implement this by getting the current velocity from the Kinematics and adding the acceleration to it, then using moveByVelocityAsync to reach the target velocity. However, I noticed that this method has a required duration as a parameter. My question is this; "Is there an API I can call that will move the drone in a direction until a target velocity is reached?"

### Include context on what you are trying to achieve

#### Context details
 Windows, Airsim v1.6.0, Python 3.7, Unreal 4.26.2
 Standard PX4 HITL settings

### Include details of what you already did to find answers
I have been reading the API Documentation reference, but the functionalities of some of the APIs were not clear to me.

jonyMarino, 2021-11-24:
@Gastastrophe, if you don't call to `.join()`, you can avoid the desired duration, and the following command will override the one that is being executed.

Gastastrophe, 2021-11-30:
> @Gastastrophe, if you don't call to `.join()`, you can avoid the desired duration, and the following command will override the one that is being executed (at least with SimpleFlight it works that way, not sure for PX4, but you can use mavsdk also)

Hmm... That might work. I can just check the velocity in a loop afterwards to see if the target velocity has been reached. Thanks

closed 2021-11-30 by Gastastrophe