microsoft/AirSim #4178: MoveToPositionAsync Problem

nicklhh, 2021-11-24:
## Question
### What's your question?

hello i am using airsim drone api via movetopositionasync. (UNITY)
I made loop movetopositionasync to move drone to decided path by 1 value of current position(path.txt).
But when I run it, if I order the drone to move only forwardly(0,1,0), of course, it moves forward correctly, but at the same time, it results in the drone moving in the z-axis (upward) direction by less than 1 value.
**Picture below --> my current quadcopter setting**
[image]

How can i solve this prb(make drone to move only on the given axis.)?
### Include context on what you are trying to achieve

// Copyright (c) Microsoft Corporation. All rights reserved.
// Licensed under the MIT License.

#include "common/common_utils/StrictMode.hpp"
STRICT_MODE_OFF
#ifndef RPCLIB_MSGPACK
#define RPCLIB_MSGPACK clmdep_msgpack
#endif // !RPCLIB_MSGPACK
#include "rpc/rpc_error.h"
STRICT_MODE_ON

#include "vehicles/multirotor/api/MultirotorRpcLibClient.hpp"
#include "common/common_utils/FileSystem.hpp"
#include <iostream>
#include <chrono> 
#include <fstream> 

using namespace std;

int main()
{
    int path_ter[5000][3] = { 0, 0, 0 };

    ifstream is;
    is.open("path.txt");
    int j = 0;

    while (j < 2000) {
        int path = 0;
        is >> path;
        path_ter[j][0] = (path / 9) - 1;
        path_ter[j][1] = ((path % 9) / 3) - 1;
        path_ter[j][2] = (path % 3) - 1;
        j++;
    }
    is.close();

    msr::airlib::MultirotorRpcLibClient client;

    cout << "Press Enter to enable API control" << endl;
    cin.get();
    client.enableApiControl(true);

    cout << "Press Enter to arm the drone" << endl;
    cin.get();
    client.armDisarm(true);

    cout << "Press Enter to takeoff" << endl;
    cin.get();
    client.takeoffAsync(5)->waitOnLastTask();
    auto position = client.getMultirotorState().getPosition();
    float z = position.z();
    const float speed = 10.0f;
    const float size = 100.0f;
    const float duration = size / speed;
    DrivetrainType drivetrain = DrivetrainType::ForwardOnly;
    YawMode yaw_mode(true, 0);
    client.moveByVelocityZAsync(0, 0, z - 3.554f, duration, drivetrain, yaw_mode);

    cout << "Press Enter to move drone" << endl;
    cin.get();

    for (int k = 0; k < j - 1; k++) {
        printf("%d %d %d \n", path_ter[k][0], path_ter[k][1], path_ter[k][2]);
        auto positionk = client.getMultirotorState().getPosition(); // from current location
        client.moveToPositionAsync(positionk.x() - path_ter[k][1], positionk.y() - path_ter[k][0], positionk.z() - path_ter[k][2], 1)->waitOnLastTask();
    }

    cout << "Press Enter to land" << endl;
    cin.get();
    client.moveToZAsync(33.71f, 5)->waitOnLastTask();

    return 0;
}
path.txt --> 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 16 7 16 16 16 7 16 16 7 16 16 7 16 16 7 16 16 16 7 16 16 7 16 16 7 16 16 16 7 16 16 7 16 16 7 16 16 7 16 16 16 7 16 16 7 16 16 7 16 16 7 16 16 16 7 16 16 7 16 16 7 16 16 7 16 16 16 7 16 16 7 16 16 7 16 16 7 16 16 16 7 16 16 7 16 16 7 16 16 16 7 16 16 7 16 16 7 16 16 7 16 16 16 7 16 16 7 16 16 7 16 16 7 16 16 16 7 16 16 7 16 16 7 16 16 7 16 16 16 7 16 16 7 16 16 7 16 16 16 7 16 16 7 16 16 7 16 16 7 16 16 16 7 16 16 7 16 16 7 16 16 7 16 16 16 7 16 16 7 16 16 7 16 16 7 16 16 16 7 16 16 7 16 16 7 16 16 16 7 16 16 7 16 16 7 16 16 7 16 16 16 7 16 16 7 16 16 7 16 16 16 7 16 16 7 16 16 7 16 16 16 7 16 6 16 16 6 16 15 15 7 15 26 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 16 16 16 16 16 16 16 16 16 16 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 22 22 25 22 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 26 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 16 14 14 14 14 14 14 14 14 14 14 14 14 14 14 14 14 14 14 26 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 25 22 25 22 22 22 25 22 22 25 22 22 25 22 22 25 22 22 22 25 22 22 25 22 22 25 22 22 22 25 22 22 25 22 22 22 25 22 22 25 22 22 25 22 22 22 25 22 22 24 22 22 24 22 22 24 22 24 22 22 24 22 24 22 22 24 22 22 24 22 21 24 22 21 22 24 21 22 21 24 22 21 21 25 21 2 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 10 10 10 1 10 10 10 10 10 19 10 10 1 10 10 10 10 19 10 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 4 1 14 14 14 14 14 14 14 14 14 2 1 1 1 1 1 1 1 1 1 1 1 1 1 10 1 10 10 1 10 10 1 10 10 10 1 10 10 1 10 10 1 10 10 1 10 10 10 1 10 10 1 10 10 1 10 10 1 10 10 10 1 10 10 1 10 10 1 10 10 1 10 10 10 1 10 10 1 10 10 1 10 10 10 1 10 10 1 10 10 1 10 10 1 10 10 10 1 10 10 1 10 10 1 10 10 1 10 10 10 1 10 10 1 10 10 1 10 10 1 10 10 10 1 10 10 1 10 10 1 10 10 1 10 10 10 1 10 10 1 10 10 1 10 10 10 1 10 10 1 10 10 1 10 10 1 10 10 10 1 10 10 1 10 10 1 10 10 1 10 10 10 1 10 10 1 10 10 1 10 10 1 10 10 10 1 10 10 1 10 10 1 10 10 10 1 10 10 1 10 10 1 10 10 1 10 10 10 1 10 10 1 10 10 1 10 10 1 10 10 10 1 10 10 1 10 10 1 10 10 1 10 10 10 1 10 10 1 10 10 1 10 10 10 1 10 10 1 10 10 1 10 10 1 10 10 10 1 10 10 1 10 10 1 10 10 1 10 10 10 1 10 10 1 10 10 1 10 10 1 10 10 10 1 10 10 1 10 10 1 10 10 10 1 10 10 1 10 10 1 10 10 1 10 10 10 1 10 10 1 10 10 1 10 10 1 10 10 10 1 10 10 1 10 10 1 10 10 1 10 10 10 1 10 10 1 10 10 1 10 10 10 1 10 10 1 10 10 1 10 10 1 10 10 10 1 10 10 1 10 10 1 10 10 1 10 10 10 1 10 10 1 10 10 1 10 10 10 1 10 10 1 10 10 1 10 10 1 10 10 10 1 10 10 1 10 10 1 10 10 1 10 10 10 1 10 10 1 10 10 1 10 10 10 1 10 10 1 10 10 1 10 10 10 1 10 10 1 10 10 1 10 10 1 10 10 10 1 10 10 1 10 10 10 0 10 0 10 10 0 10 0 10 10 9 0 10 9 9 0 10  

#### Context details

Windows 10 20H2, Unity 2020.3.8f1

{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor"
}

### Include details of what you already did to find answers

jonyMarino, 2021-11-24:
@nicklhh. There is a lookahead error that you should take into account. The moveToPosition() API moves the drone to a position within a specific volume whose center is the target position. I think you should correct that error inside the arguments that you pass to moveToPosition(). Also, incrementing the position by one can lead to a great relative error (because of the same lookahead error).

nicklhh, 2021-11-25:
> @nicklhh. There is a lookahead error that you should take into account. The moveToPosition() API moves the drone to a position within a specific volume whose center is the target position. I think [...]

ah, i understand. thx a lot :(

closed 2021-11-25 by nicklhh

Cming7, 2022-01-15:
> @nicklhh. There is a lookahead error that you should take into account. The moveToPosition() API moves the drone to a position within a specific volume whose center is the target position. I think [...]

What's the lookahead error?