microsoft/AirSim #4196: Object Detection bounding boxes are not returning valid results in AirSim CityEnviron

tkupek, 2021-12-06:
## Bug report 

- AirSim Version/#commit: v.1.6.0 Windows
- UE/Unity version: PCD3D_SM5
- autopilot version: -
- OS Version: Windows 10

### What's the issue you encountered?

I am using the tutorial for Object Detection (https://microsoft.github.io/AirSim/object_detection/#usage-example) to generate groundtruth data for pedestrians. I set the filter mesh to `*` to see all objects:
`client.simAddDetectionFilterMeshName(camera_name, image_type, "*")`

The returned boxes seem weird and are definitely not correct.
[image]

What am I doing wrong? Did I choose the wrong camera? Why are no detections of pedestrians showing up?

### Settings

```
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "ComputerVision",
  "CameraDefaults": {
      "CaptureSettings": [
        {
          "ImageType": 0,
          "Width": 1280,
          "Height": 720,
          "FOV_Degrees": 90,
          "AutoExposureSpeed": 100,
          "MotionBlurAmount": 0
        }
    ]
  }
}
```

### How can the issue be reproduced?

1. Use PythonClient detection example with AirSim CityEnviron

jonyMarino, 2021-12-13:
Hi and welcome @tkupek! Can you try to detect just the pedestrians? You can find the with `simListSceneObjects()`

tkupek, 2021-12-14:
Thanks for the response. The list of objects has over 5000 entries. I cannot find any with "pedestrian" or "person".

[airsim-city-objects.txt](https://github.com/microsoft/AirSim/files/7711859/airsim-city-objects.txt)

jonyMarino, 2021-12-14:
I am afraid that this environment will not serve you for what you need. All pedestrians are parts of CityEnv_PedManager_26 so you cannot individualize them. You are going to need a custom environment.

tkupek, 2021-12-15:
That's too bad. Pedestrians and cyclist are the most important classes for me.

jonyMarino, 2021-12-15:
Yes, that would be a good feature. But we can't work on this soon. So if you find a way to make, that work let us know how we should implement it for the City environment.

zimmy87, 2022-01-05:
Now that #4270 is merged, the boxes for the buildings and other actors should be more accurate in the next release. However, like @jonyMarino mentioned earlier, the pedestrians, cyclists, and cars are all part of instanced static mesh components inside various manager actors. The current object detection algorithm doesn't make individual detections for instanced static mesh components, so you won't see any detections for these objects.

chongjeelee, 2022-01-25:
I find a problem,when i set the object moving ,i can not detect very accurate position(just the moving object's trace area),it's very big!
below picture show the result.the first picture is detecting moving air_balloon,the second picture is detecting static air_balloon.
[image]
----
[image]
---
so ,how can i get the moving object's 2d-box on time?