microsoft/AirSim #4197: LogPackageName: SearchForPackageOnDisk took   0.008s, but failed to resolve Blocks

debian-user-france1, 2021-12-06:
## Bug report 

- AirSim Version/#commit: 1.6.0 prebuilt: "Africa, Abandoned Park, ..."
- UE/Unity version: N/A
- autopilot version: N/A
- OS Version: Manjaro: `Linux Manjaro-Desktop 5.15.2-2-MANJARO #1 SMP PREEMPT Sat Nov 13 19:25:38 UTC 2021 x86_64 GNU/Linux`

### What's the issue you encountered?

I start it: ./NAME.sh
The screen gets black (two windows opened, one 'select vehicle' and the other has a longer name.
Once it's killed by ctrl+windows+alt F4 -> `killall NAME` eg. Africa_001 all is okay and I see following error messages:
```
[2021.12.06-16.16.45:088][  0]LogStreaming: Error: Couldn't find file for package /Engine/EditorMaterials/AssetViewer/EpicQuadPanorama_CC+EV1 requested by async loading code. NameToLoad: /Engine/EditorMaterials/AssetViewer/EpicQuadPanorama_CC+EV1
[2021.12.06-16.16.45:088][  0]LogStreaming: Error: Found 1 dependent packages...
[2021.12.06-16.16.45:088][  0]LogStreaming: Error:   /AirSim/VehicleAdv/SUV/AutomotiveMaterials/Materials/Glass/M_Glass
[2021.12.06-16.16.45:119][  0]LogAIModule: Creating AISystem for world FlyingExampleMap
[2021.12.06-16.16.45:119][  0]LogAISub: UAISubsystem EnvQueryManager (0x7f943fe38800), frame # 0
[2021.12.06-16.16.45:119][  0]LogAISub: UAISubsystem AIPerceptionSystem (0x7f948626e840), frame # 0
[2021.12.06-16.16.45:119][  0]LogLoad: Game class is 'AirSimGameMode'
[2021.12.06-16.16.45:121][  0]LogWorld: Bringing World /Game/FlyingCPP/Maps/FlyingExampleMap.FlyingExampleMap up for play (max tick rate 0) at 2021.12.06-17.16.45
[2021.12.06-16.16.45:121][  0]LogWorld: Bringing up level for play took: 0.001165
[2021.12.06-16.16.45:127][  0]LogPackageName: SearchForPackageOnDisk took   0.005s, but failed to resolve Blocks.
FUnixPlatformMisc::RequestExitWithStatus
FUnixPlatformMisc::RequestExit
FUnixPlatformMisc::RequestExitWithStatus
FUnixPlatformMisc::RequestExit
FUnixPlatformMisc::RequestExit
FUnixPlatformMisc::RequestExit
```
And I see` [2021.12.06-16.16.43:081][  0]LogHAL: Warning: Splash screen image not found.`

### Settings
DEFAULT

### How can the issue be reproduced?
N/A 

### Include full error message in text form
```
    ~/Dokumente/FlySim/Blocks  cd LinuxNoEditor                                                                                                           ✔ 
    ~/Dokumente/FlySim/Blocks/LinuxNoEditor  ls                                                                                                           ✔ 
Blocks  Blocks.sh  Engine  Manifest_DebugFiles_Linux.txt  Manifest_NonUFSFiles_Linux.txt  Manifest_UFSFiles_Linux.txt
    ~/Dokumente/FlySim/Blocks/LinuxNoEditor  ./Blocks.sh                                                                                                  ✔ 
Increasing per-process limit of core file size to infinity.
- Existing per-process limit (soft=18446744073709551615, hard=18446744073709551615) is enough for us (need only 18446744073709551615)
LogPlatformFile: Using cached read wrapper
LogTaskGraph: Started task graph with 5 named threads and 14 total threads with 3 sets of task threads.
LogStats: Stats thread started at 0.030796
LogICUInternationalization: ICU TimeZone Detection - Raw Offset: +1:00, Platform Override: ''
LogPluginManager: Mounting plugin ImgMedia
LogPluginManager: Mounting plugin MediaCompositing
LogPluginManager: Mounting plugin WmfMedia
LogPluginManager: Mounting plugin WebMMedia
LogPluginManager: Mounting plugin AvfMedia
LogPluginManager: Mounting plugin LauncherChunkInstaller
LogPluginManager: Mounting plugin NetcodeUnitTest
LogPluginManager: Mounting plugin AnimationSharing
LogPluginManager: Mounting plugin UObjectPlugin
LogPluginManager: Mounting plugin UdpMessaging
LogPluginManager: Mounting plugin TcpMessaging
LogPluginManager: Mounting plugin ChaosClothEditor
LogPluginManager: Mounting plugin GeometryCache
LogPluginManager: Mounting plugin ChaosEditor
LogPluginManager: Mounting plugin ChaosNiagara
LogPluginManager: Mounting plugin GeometryCollectionPlugin
LogPluginManager: Mounting plugin AutomationUtils
LogPluginManager: Mounting plugin BackChannel
LogPluginManager: Mounting plugin GeometryProcessing
LogPluginManager: Mounting plugin PlatformCrypto
[... 720 lines cut ...]
XYXYXY XYXYXY Clearnupworld 0x7f943e876140 -> (nil)
[2021.12.06-16.16.45:076][  0]LogWorld: UWorld::CleanupWorld for Untitled, bSessionEnded=true, bCleanupResources=true
[2021.12.06-16.16.45:076][  0]LogSlate: InvalidateAllWidgets triggered.  All widgets were invalidated
[2021.12.06-16.16.45:084][  0]LogUObjectHash: Compacting FUObjectHashTables data took   0.55ms
[2021.12.06-16.16.45:088][  0]LogStreaming: Error: Couldn't find file for package /Engine/EditorMaterials/AssetViewer/EpicQuadPanorama_CC+EV1 requested by async loading code. NameToLoad: /Engine/EditorMaterials/AssetViewer/EpicQuadPanorama_CC+EV1
[2021.12.06-16.16.45:088][  0]LogStreaming: Error: Found 1 dependent packages...
[2021.12.06-16.16.45:088][  0]LogStreaming: Error:   /AirSim/VehicleAdv/SUV/AutomotiveMaterials/Materials/Glass/M_Glass
[2021.12.06-16.16.45:119][  0]LogAIModule: Creating AISystem for world FlyingExampleMap
[2021.12.06-16.16.45:119][  0]LogAISub: UAISubsystem EnvQueryManager (0x7f943fe38800), frame # 0
[2021.12.06-16.16.45:119][  0]LogAISub: UAISubsystem AIPerceptionSystem (0x7f948626e840), frame # 0
[2021.12.06-16.16.45:119][  0]LogLoad: Game class is 'AirSimGameMode'
[2021.12.06-16.16.45:121][  0]LogWorld: Bringing World /Game/FlyingCPP/Maps/FlyingExampleMap.FlyingExampleMap up for play (max tick rate 0) at 2021.12.06-17.16.45
[2021.12.06-16.16.45:121][  0]LogWorld: Bringing up level for play took: 0.001165
[2021.12.06-16.16.45:127][  0]LogPackageName: SearchForPackageOnDisk took   0.005s, but failed to resolve Blocks.
FUnixPlatformMisc::RequestExitWithStatus
FUnixPlatformMisc::RequestExit
FUnixPlatformMisc::RequestExitWithStatus
FUnixPlatformMisc::RequestExit
FUnixPlatformMisc::RequestExit
FUnixPlatformMisc::RequestExit
```

**What's better than filing an issue? Filing a pull request :).**  

DUDE Github is crashing I can't see what I'm writing

debian-user-france1, 2021-12-06:
#4068 (Issue where I've put short information but I thought it's not related and is closed)

debian-user-france1, 2021-12-06:
I discovered this newly: [VIDEO LINK](https://cloud.privacyy.ch/index.php/s/bqLw4owByMZRNat)

debian-user-france1, 2021-12-06:
I can steer Africa now with the arrow keys (car)

jonyMarino, 2021-12-13:
Hi and welcome @debian-user-france1! AirSim vehicles don't drive by themselves. However, AirSim provides an API that you can use to accomplish that.