microsoft/AirSim #4231: Build error after airsim update

Sabrin0, 2021-12-17:
I followed the instruction  [Upgrading Your Custom Unreal Project](https://microsoft.github.io/AirSim/unreal_upgrade/#upgrading-your-custom-unreal-project), but now I'm no more able to run the simulation, both by building in Visual Studio (first image error) or by launching the .uproject

[image]
[image]
( the version is the same as before: UE4.26)
[image]

_Originally posted by @Sabrin0 in https://github.com/microsoft/AirSim/discussions/4214#discussioncomment-1833997_

Sabrin0, 2021-12-17:
In order to solve the firs error I had to set as startup project the game from the VS solution explorer. By the way, new errors jumped out:

Build started...
1>------ Build started: Project: DroneRescueNav, Configuration: DebugGame_Editor x64 ------
1>Using 'git status' to determine working set for adaptive non-unity build (C:\Users\Luca\Desktop\University\Thesis\DroneRescueNav).
1>Building DroneRescueNavEditor...
1>Using Visual Studio 2019 14.29.30038 toolchain (C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30037) and Windows 10.0.19041.0 SDK (C:\Program Files (x86)\Windows Kits\10).
1>[Adaptive unity build] Excluded from AirSim unity file: AirBlueprintLib.cpp, RpcLibClientBase.cpp, RpcLibServerBase.cpp, MultirotorApiBase.cpp, MultirotorRpcLibClient.cpp, MultirotorRpcLibServer.cpp, CameraDirector.cpp, PawnSimApi.cpp, PIPCamera.cpp, SimHUD.cpp, SimModeBase.cpp, SimModeWorldBase.cpp, CarPawnSimApi.cpp, SimModeCar.cpp, ComputerVisionPawn.cpp, SimModeComputerVision.cpp, MultirotorPawnSimApi.cpp, WorldSimApi.cpp
1>Building 4 actions with 12 processes...
1>  [1/4] WorldSimApi.cpp
1>C:\Users\Luca\Desktop\University\Thesis\DroneRescueNav\Plugins\AirSim\Source\WorldSimApi.cpp(970): error C2653: 'AirSimSettings': is not a class or namespace name
1>C:\Users\Luca\Desktop\University\Thesis\DroneRescueNav\Plugins\AirSim\Source\WorldSimApi.cpp(970): error C3861: 'singleton': identifier not found
1>C:\Users\Luca\Desktop\University\Thesis\DroneRescueNav\Plugins\AirSim\Source\WorldSimApi.cpp(969): error C2660: 'msr::airlib::EarthUtils::nedToGeodetic': function does not take 1 arguments
1>  C:\Users\Luca\Desktop\University\Thesis\DroneRescueNav\Plugins\AirSim\Source\AirLib\include\common/EarthUtils.hpp(291): note: see declaration of 'msr::airlib::EarthUtils::nedToGeodetic'
1>C:\Users\Luca\Desktop\University\Thesis\DroneRescueNav\Plugins\AirSim\Source\WorldSimApi.cpp(972): error C3861: 'Vector2r': identifier not found
1>C:\Users\Luca\Desktop\University\Thesis\DroneRescueNav\Plugins\AirSim\Source\WorldSimApi.cpp(973): error C3861: 'Vector2r': identifier not found
1>C:\Users\Luca\Desktop\University\Thesis\DroneRescueNav\Plugins\AirSim\Source\WorldSimApi.cpp(979): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
1>C:\Users\Luca\Desktop\University\Thesis\DroneRescueNav\Plugins\AirSim\Source\WorldSimApi.cpp(979): error C2143: syntax error: missing ';' before '&'
1>C:\Users\Luca\Desktop\University\Thesis\DroneRescueNav\Plugins\AirSim\Source\WorldSimApi.cpp(979): error C2065: 'orientation': undeclared identifier
1>C:\Users\Luca\Desktop\University\Thesis\DroneRescueNav\Plugins\AirSim\Source\WorldSimApi.cpp(981): error C2065: 'orientation': undeclared identifier
1>C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.MakeFile.Targets(45,5): error MSB3073: The command ""C:\Program Files\Epic Games\UE_4.26\Engine\Build\BatchFiles\Build.bat" DroneRescueNavEditor Win64 DebugGame -Project="C:\Users\Luca\Desktop\University\Thesis\DroneRescueNav\DroneRescueNav.uproject" -WaitMutex -FromMsBuild" exited with code 6.
1>Done building project "DroneRescueNav.vcxproj" -- FAILED.
========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========

XueZhihan, 2021-12-20:
Same problem. Have you solved it now？

zimmy87, 2021-12-21:
I'm not able to repro this personally by following the update steps on the Blocks project w/ either UE 4.25 or UE 4.26. Can you provide a few more details:

1. is Plugins\AirSim\Source\AirLib\include\common\AirSimSettings.hpp present in your project directory structure?
1. can you share any log files in Saved\logs?
1. when performing step 3 (replacing the plugin), did you overwrite an existing old copy of the plugin or did you delete the old copy before placing the new one?
1. is the source of your custom environment available anywhere online?

Sabrin0, 2021-12-24:
HI, thank you for your support. Yes the file `Plugins\AirSim\Source\AirLib\include\common\AirSimSettings.hpp` is still present. Regarding the 2., I've tried in both ways. I attached the last log file.
For the 4. I've just a private report.

[UnrealVersionSelector-2021.12.24-16.25.04.log](https://github.com/microsoft/AirSim/files/7774706/UnrealVersionSelector-2021.12.24-16.25.04.log)

Sabrin0, 2021-12-27:
@XueZhihan did you solve it?

alonfaraj, 2021-12-28:
Hi @Sabrin0 
I see that you have git control for your project.
Can you try change the folder `.git` to anything else (`.git_`), and then rebuild?

Sabrin0, 2021-12-29:
Hi @alonfaraj, thanks for tour reply. I saw it too by looking at the log file. Git's given my build troubles, I don't know why. In order to solve the problem I modified the [BuildConfiguration.xml](https://docs.unrealengine.com/4.26/en-US/ProductionPipelines/BuildTools/UnrealBuildTool/BuildConfiguration/) by adding the following lines:

 ```
     <SourceFileWorkingSet> 
          <Provider>None</Provider> 
          <RepositoryPath></RepositoryPath> 
          <GitPath></GitPath> 
      </SourceFileWorkingSet>
```

Now everything works properly. Is this a good solution?

alonfaraj, 2021-12-29:
I'm not familiar with this file and personally I prefer not change UE4 related files. 
My idea is to disable git by change its folder name temporary, build and rename it again to the original name.
Glad to hear it's works for you!

closed 2021-12-29 by Sabrin0