microsoft/AirSim #4235: No instance of VS Error.

Mohamed-Omran, 2021-12-22:
## Question
### What's your question?

I cannot build AirSim with VS 2022.

### Include context on what you are trying to achieve

When executing `build.cmd` I get the following error:
```
CMake Error:
  Generator

    Visual Studio 16 2019

  could not find any instance of Visual Studio.
```
#### Context details

Win 10
VS 17.0 
UE 4.27.2

### Include details of what you already did to find answers
N/A

alonfaraj, 2021-12-23:
AirSim currently support VS 2019.
You can try change this line in `build.cmd`:
`cmake -G"Visual Studio 16 2019" ..`
to:
`cmake -G"Visual Studio 17 2022" ..`
but I'm not sure it's going to work

Mohamed-Omran, 2021-12-25:
Thank you @alonfaraj.
I tried the version this amazing person @elevir forked and it worked fine on VS 2022

closed 2021-12-26 by Mohamed-Omran

rhalaly, 2021-12-29:
@Mohamed-Omran can you share more info on how to build it with VS 2022? Which fork did you use?

Mohamed-Omran, 2022-01-07:
@rhalaly  https://github.com/elevir/AirSim