microsoft/AirSim #4241: clockspeed seems doesn't work

ZhangFanSC, 2021-12-27:
## Question
### What's your question?

#### Context details

[image]

[image]

### Include details of what you already did to find answers
How do I set up clockSpeed to work？@ahmed-elsaharti  Thanks.

ahmed-elsaharti, 2021-12-27:
Hi @ZhangFanSC , From what I'm seeing in the code https://github.com/microsoft/AirSim/blob/bb9922eb0d56877d8e987ec1c24acc13c4ab3730/Unreal/Plugins/AirSim/Source/Vehicles/Multirotor/MultirotorPawnSimApi.cpp#L85 it looks like the timestamp is independent of the simulated clock speed and basically reports back [the time since epoch ](https://microsoft.github.io/AirSim/apis/#getmultirotorstate) regardless of the simualtion's clock and/or if the sim is paused.

ZhangFanSC, 2021-12-28:
So my idea is that if I set clockSpeed to 4, and I change the time interval between step execution in the corresponding reinforcement learning to 1/4 as before, will that work in the end?
@ahmed-elsaharti 
Thanks.

ahmed-elsaharti, 2021-12-28:
@ZhangFanSC if im understanding what you're saying correctly, doing this will result in a 'step execution' every '1 simulated second'.
Since 1 real world second = 4 simulated seconds, A command executed every 1/4 real world second should result in a that command being executed every 1 'simulated second'.... i think xD

ZhangFanSC, 2021-12-28:
ok，thank you.@ahmed-elsaharti

jonyMarino, 2021-12-31:
This is related to #4232.

Thanks, @ahmed-elsaharti!! Good to see you again!

ahmed-elsaharti, 2022-01-07:
Thanks @jonyMarino! I try to take a glance whenever possible 🤓

zimmy87, 2022-01-20:
Hi @ZhangFanSC, I've put out #4305 for fixing this behavior, please test with the PR applied and let me know if you have any issues

closed 2022-01-31 by jonyMarino

haotiangu, 2023-10-18:
@zimmy87  Hi, I am trying to change the clock speed from 1 to 0.5 in setting file. Seems like the test output is still default clockspeed when running the test code posed by @jonyMarino . Could you guys please give me some ideas. The airsim is the latest version. UE IS 4.27.1. Ubuntu 18.04
haotian@haotian-911-SE-KN:~/AirSim/PythonClient/car$ python3 clockspeed.py
Connected!
Client Ver:1 (Min Req: 1), Server Ver:1 (Min Req: 1)

1697666955089885000
0.0
1.00804
1.001178
0.997832
1.006489
0.999038
1.001592
0.99849
^CTraceback (most recent call last):
  File "clockspeed.py", line 23, in <module>
    time.sleep(1.0)
KeyboardInterrupt

PR #4305 (2022-01-31): fix clock speed in car mode

Fixes: #4241    
Fixes: #4232

## About
The car mode currently utilizes a default clock for returning the CarState's timestamp, so the clock speed from the settings isn't reflected in the timestamp. This change adds logic for initializing the appropriate clock type from settings as well as logic for reading the timestamp from the newly initialized clock. 

## How Has This Been Tested?
The following script was used in car mode to confirm the clock was advancing at the proper clock speed:

```
import setup_path 
import airsim

import pprint
import time

# connect to  [... cut]