microsoft/AirSim #4242: help on how to use my car instead of default car in airsim

dellhuyongcai, 2021-12-27:
## Question
### What's your question?

I just want to learn how to use  my car in airsim, no documents can be referred,just come here to find materials,hope someone give steps as clearly as possible,thanks a lot!!

### Include context on what you are trying to achieve

#### Context details

### Include details of what you already did to find answers

Sabrin0, 2021-12-27:
Are you using Unreal? In this case first of all you need to declare the pawn path in the `setting.json`. For example:

```
"PawnPaths": {
    "MyVehicle": { "PawnBP": "Class'/Game/path/to/BP_MyVehicle.BP_MyVehicle_C'" }
  },
```
Then  for the vehicle setting:

```
"Vehicles": {

    "MyVehicle": {
      .
      .
      "PawnPath": "MainDrone",
      .
      .
      },
      }
```
I did so and It works properly. Of course you must have your car BP properly configured in UE (e.g. mesh, material, actuators...)

dellhuyongcai, 2022-01-04:
thanks a lot,Successfully solved this problem!

closed 2022-01-04 by dellhuyongcai