microsoft/AirSim #4243: WorldSimApi. cpp build error

Sabrin0, 2021-12-27:
Whenever I build the project with visual studio I get the following errors referred to the file `Plugins\AirSim\Source\WorldSimApi.cpp` :

```
WorldSimApi.cpp(1052): error C2653: 'AirSimSettings': is not a class or namespace name
WorldSimApi.cpp(1052): error C3861: 'singleton': identifier not found
WorldSimApi.cpp(1051): error C2660: 'msr::airlib::EarthUtils::nedToGeodetic': function does not take 1 arguments
\Plugins\AirSim\Source\AirLib\include\common/EarthUtils.hpp(291): note: see declaration of 'msr::airlib::EarthUtils::nedToGeodetic'
WorldSimApi.cpp(1054): error C3861: 'Vector2r': identifier not found
WorldSimApi.cpp(1055): error C3861: 'Vector2r': identifier not found
WorldSimApi.cpp(1061): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
WorldSimApi.cpp(1061): error C2143: syntax error: missing ';' before '&'
WorldSimApi.cpp(1061): error C2065: 'orientation': undeclared identifier
WorldSimApi.cpp(1063): error C2065: 'orientation': undeclared identifier
```
---------------------------------------------------------------------------------------------------------------------------

- Windows 10
- UE4.26
- Visual Code 2019 [Debug Game Editor Win64]
- Windows 10.0.19041.0 SDK
- .NET 5.0

Sabrin0, 2021-12-28:
Maybe this warning might be correlated:

```
WARNING: Library 'Shell32.lib' was not resolvable to a file when used in Module 'AirSim', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
WARNING: Library 'dinput8.lib' was not resolvable to a file when used in Module 'AirSim', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
WARNING: Library 'dxguid.lib' was not resolvable to a file when used in Module 'AirSim', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
```

Sabrin0, 2021-12-29:
#4231

closed 2021-12-29 by Sabrin0

andrewjong, 2022-10-11:
Did you figure out how to solve this?

andrewjong, 2022-10-14:
@Sabrin0

xxEoD2242, 2022-10-14:
Hey @andrewjong ,

You need to upgrade to Visual Studio 2022 and make sure to remove the old Windows SDK (there is a new version available). Also, you need to update to UE 4.27.2.

Also, MicroSoft has shutdown this open source project. We are continuing that on here: [Colosseum](https://github.com/CodexLabsLLC/Colosseum)

Sabrin0, 2022-10-15:
Hi @andrewjong, can you kindly share the log files in Saved\logs?