microsoft/AirSim #4247: lidar sensor can not get the real data from environment

chongjeelee, 2021-12-28:
## Bug report 
N/A

### What's the issue you encountered?

### Settings

### How can the issue be reproduced?

1.
2.

### Include full error message in text form

**What's better than filing an issue? Filing a pull request :).**

-----------------
hi,when i use lidar sensor to get 3D data from the environment(TrapCam or MSbuilding2018),i find the 3D data can not include any object ,it like this below,the right part is ground 3d data,but can not find the animal information. i think the strange result is because of the UE4 environment ,maybe the environment can not include any physical parameters on the object(for instance,animal,grass,stone and so on.),it just works well on vision part,not for reflective and correct collision part.
So,when someone want to use those environment for 3D detecting ,it dose not work any more!
i think this is a big bug...right?
[image]

chongjeelee, 2021-12-28:
put lidar segmentation info to point cloud ([x y z intensity])
---------
[image]

jonyMarino, 2021-12-31:
Hi @chongjeelee! The Lidar depends on how the collisions have been configured. Take a look at #2948. Let us see which is the collisions configuration for that objects.

zimmy87, 2022-01-13:
The collision presets on various objects in TrapCam and MSbuilding2018 have been updated to respond to Lidar. For custom environments, objects need to block the visibility collision channel in order to be detected by lidar.

closed 2022-02-11 by chongjeelee