microsoft/AirSim #4250: What coordinates of radar in SensorLocalFrame？

ZhangFanSC, 2021-12-29:
## Question
### What's your question?

1. For radar, I set the DataFrame to SensorLocalFrame. I would like to ask if the coordinates obtained at this time are relative to the local NED coordinate system of the vehicle starting point or the liDAR's own coordinate system (which is converted from the local vehicle coordinate system to the sensor coordinate system through coordinate rotation)? If it's the latter, does that mean that the sensor's axes coincide with the vehicle's direction of travel? So I can get the liDAR data right in front of the vehicle

2. Where does liDAR data start to scan? For example, if I know a single two-dimensional lidar plane, where does it scan from in the vehicle (0,360)?

### Include context on what you are trying to achieve

#### Context details

[image]

### Include details of what you already did to find answers
The first question is what is the coordinate system for Lidar under SensorLocalFrame?

The second question is what is the direction of the starting point of lidar's data?

@Ratnesh Madaan

Sabrin0, 2021-12-29:
Due to the [LiDAR ](https://microsoft.github.io/AirSim/lidar/)parameters, configured via the `setting.json`, you can define the position (NED [m]) and the orientation (order to front vector +X [deg]) of the sensor wrt the vehicle frame.  Lidar provides also a `point_cloud`, a flat array of floats along with the timestamp of the capture and lidar pose. The floats represent the x,y,z for each point hit within the range in the last scan. These points can be wrt the **SensorLocalFrame** or the **VehicleInertialFrame**. 

Regarding the second question, the scan start position depends on the parameter `HorizontalFOVStart` while the end on  `HorizontalFOVEnd` (both in degrees).

I suggest you to set as true the flag `DrawsDebugPoints`, it requires a huge memory usage but it allows you to figure out the lidar configuration

ZhangFanSC, 2021-12-30:
Thank you for solving my second problem.

But I still have a question about the first one.Suppose my liDAR settings is as follows：
[image]

My first question is foucs on in sensorLocalFrame what coordinate system did I get the data from the point cloud .As the figure below.
[image]
The left is for the vehicle coordinate system, Lidar's coordinate system does not make any rotation changes, only shifts.
The right is the liDAR coordinate system and the forward direction of the vehicle coincide, that is, the LiDAR coordinate system is rotated and shifts through the vehicle coordinate system.

I think the correct one is the one on the right, but I'm not sure. I hope you can give me some suggestions.@Sabrin0 Thanks

Ryton, 2022-02-21:
Yes, the right one. its a concatenation of the vehicle pose and the lidar /sensor pose (xyz+yaw+pitch+roll).
You can get an output of the transformations in a rosbag.

PS: better change the orizontal field of view to -90 to +90, then your (sensor-X-axis) is the front of your lidar/radar senors.