microsoft/AirSim #4251: How do I replace the quaternion data in airsim_rec.txt with pitch, yaw, & roll angle?

adithyashok, 2021-12-29:
## Question
### What's your question?

Apologies in advance if this question is trivial. I am attempting to generate a dataset for training a DNN, and was wondering if there is any way to change the data that is recorded into airsim_rec.txt by default? I was hoping instead of Q_X, Q_Y, Q_Z, Q_W to record the pitch, roll, and yaw of the drone. 

### Include context on what you are trying to achieve

N/A
#### Context details

Running on linux, Python 3.6 and the latest unreal version. 
### Include details of what you already did to find answers
I've looked through this, but haven't found sufficient information to get things working 
https://github.com/microsoft/AirSim/blob/master/docs/modify_recording_data.md

If I'm not wrong, this section of code needs to be changed: https://github.com/microsoft/AirSim/blob/880c5541fd4824ee2cd9bb82ca5f611eb1ab236a/Unreal/Plugins/AirSim/Source/PawnSimApi.cpp#L556

I'm not sure how to convert the orientation to pitch/roll/yaw without python, which is what I've been working with thus far.

alonfaraj, 2021-12-29:
Hi @adithyashok,
You can use `VectorMath::toEulerianAngle()`.
so in your case:
```
    float p, r, y;
    VectorMath::toEulerianAngle(kinematics->pose.orientation, p, r, y);
```
notice that the output `p, r, y` angels are in radians.

adithyashok, 2021-12-29:
Thanks, this looks great. Despite modifying the file, changes are not seen in airsim_rec.txt. Do I need to make these updates elsewhere as well?

alonfaraj, 2021-12-29:
Did you add p, r, y to the part that makes the actual writing?

adithyashok, 2021-12-29:
Yes, I did.

alonfaraj, 2021-12-30:
I have made a quick test with the following code:

```
std::string PawnSimApi::getRecordFileLine(bool is_header_line) const
{
    if (is_header_line) {
        return "VehicleName\tTimeStamp\tPOS_X\tPOS_Y\tPOS_Z\tQ_W\tQ_X\tQ_Y\tQ_Z\t";
    }

    static int cnt = 0;

    std::ostringstream ss;
    ss << "Hello World " << cnt  << "\t";
    cnt++;

    return ss.str();
}
```

and the output file looks like this:
```
VehicleName	TimeStamp	POS_X	POS_Y	POS_Z	Q_W	Q_X	Q_Y	Q_Z	ImageFile
Hello World 0	img_SimpleFlight__0_1640890968359529500.png
Hello World 1	img_SimpleFlight__0_1640890968418712200.png
Hello World 2	img_SimpleFlight__0_1640890968470045300.png
Hello World 3	img_SimpleFlight__0_1640890968519472200.png
Hello World 4	img_SimpleFlight__0_1640890968571873100.png
Hello World 5	img_SimpleFlight__0_1640890968624099700.png
```

I can't think of any way that you have modified the code and rebuild, but still get the old record file.
Can you share your code? or use breakpoint to check that you actually go into this function?

adithyashok, 2021-12-30:
#Looking into it, I realized that I didn't update the code in the correct plugin location for my Unreal project, and making the update does the job. Oops! Resolved.