microsoft/AirSim #4253: Cannot compile environment to Linux

rhalaly, 2021-12-29:
## Bug report 

- AirSim Version/#commit: bb9922eb0d56877d8e987ec1c24acc13c4ab3730
- UE/Unity version: 4.27
- autopilot version:
- OS Version: Windows 11

### What's the issue you encountered?

[image]

### Settings

### How can the issue be reproduced?

1. Open an environment in Windows
2. Package project to Linux / Linux (AArch64)

### Include full error message in text form
> fatal error: 'UdpSocket.hpp' file not found
>PackagingResults: Error: loop variable 'preset' creates a copy from type 'const FNamedLensPreset' [-Werror,-Wrange-loop-construct]
PackagingResults: Error: loop variable 'preset' creates a copy from type 'const FNamedFilmbackPreset' [-Werror,-Wrange-loop-construct]

rhalaly, 2022-01-03:
@bozcani Thank you for the fix, but there is also a problem that `UdpSocket.hpp` file not found

bozcani, 2022-01-03:
Ah no, my bad. I see your problem is different. I compiled a custom environment on Linux for Linux. And, I got almost the same error messages:
```
 PackagingResults: Error: loop variable 'preset' creates a copy from type 'const FNamedLensPreset' [-Werror,-Wrange-loop-construct]
PackagingResults: Error: loop variable 'preset' creates a copy from type 'const FNamedFilmbackPreset' [-Werror,-Wrange-loop-construct]
```
My PR fixes this for building on Linux for Linux but it doesn't necessarily solve your problem. I will remove the link to your issue from my PR message.

rhalaly, 2022-01-03:
@bozcani thanks for the clarification. Actually, I didn't succeed to compile Linux on Linux even for the basic Blocks environment with Unreal Engine 4.27, only with 4.25... (Fixed the `loop variable` error manually)

rhalaly, 2022-01-03:
@bozcani thanks for the clarification. Actually, I didn't succeed to compile Linux on Linux even for the basic Blocks environment with Unreal Engine 4.27, only with 4.25... (Fixed the `loop variable 'preset' creates a copy from type 'const FNamedLensPreset' [-Werror,-Wrange-loop-construct]` manually

bozcani, 2022-01-03:
Ok, I have created a new issue (https://github.com/microsoft/AirSim/issues/4264) just to solve "loop variable" and pointed my MR (https://github.com/microsoft/AirSim/pull/4262) to that. I think your problem is different (i.e., `fatal error: 'UdpSocket.hpp' file not found` ) and have no idea for the solution at the moment. I hope someone can help.