microsoft/AirSim #4255: Support for lights attached to vehicles

Davidsastresas, 2021-12-30:
I think it would be useful to be able to change the lights attached to vehicles, in order to better estimate simulations when computer vision is involved in a system where it can control lights.

I think I have sorted out how to include permanent lights attached to the player in Unreal, but I don't know how to start attaching that to a connection in Airsim, and thus to the autopilot or software managing the vehicle.

I would not mind to work on this and do a PR, but I would appreciate a couple of hints about where to look in Airsim to make the connection to such element in unreal.

Thanks!

rajat2004, 2021-12-31:
Recently a feature to control lights was added, see https://github.com/microsoft/AirSim/pull/3991, might be useful here
For a quick hack, might be possible to continuously change the pose of the light object as needed to follow the vehicle, using say `simSetObjectPose`. Better way wold probably be to attach the blueprint to the vehicle itself, but not sure if it would still work through the API

Davidsastresas, 2021-12-31:
Thank you very much, that is enough for me to keep pulling that string. If I manage to do something nice I will submit a PR about it. Thanks!