microsoft/AirSim #4261: The Position of multi-drones is not correct.

robot-xyh, 2022-01-03:
## Bug report 

- AirSim Version/#commit:1.6.0
- UE/Unity version:
- autopilot version:
- OS Version:win10 blocks

### What's the issue you encountered?

When I set up multiple uAVs, I found that no matter how I set the position of UAVs, the position obtained was the same. I thought that each UAVs had a coordinate system, instead of a global coordinate system. But unreal POS is correct.If, in accordance with the [https://microsoft.github.io/AirSim/apis/](https://microsoft.github.io/AirSim/apis/) about the content of the Coordinate System is introduced, converting the from Unreal coordinates to NED, The location of multi-uavs is incorrect. I have noticed that` The starting point of The vehicle is always coordinates (0, 0, 0) in NED System`, but this sometimes causes problems for multiple uavs
### Settings

### How can the issue be reproduced?

1.Set a drone position to `"X": 0, "Y": 0, "Z": -2,`
The other position is `"X": -50, "Y": 0, "Z": -2,`

2.But using `simGetGroundTruthKinematics` retrieves the position of the unmanned aerial vehicle (uav), the position of the two unmanned aerial vehicle (uav) is the same

### Include full error message in text form
>>> `print(client.simGetGroundTruthEnvironment("Drone1").position)`
<Vector3r> {   'x_val': 0.0,
    'y_val': 0.0,
    'z_val': 2.6591591835021973}
>>>` print(client.simGetGroundTruthEnvironment("Drone0").position)`
<Vector3r> {   'x_val': 0.0,
    'y_val': 0.0,
    'z_val': 2.6591591835021973}
[image]

**What's better than filing an issue? Filing a pull request :).**

alonfaraj, 2022-01-07:
Related:
https://github.com/microsoft/AirSim/discussions/3844
https://github.com/microsoft/AirSim/issues/3882

zimmy87, 2022-01-11:
Looking at the code, this appears to be by design. If you take a look at the implementation of [PawnSimApi::getPose()](https://github.com/microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/PawnSimApi.cpp#L426), it converts the pawn's global position to a local offset from the pawn's starting point. This method is used as the basis for the position returned by both `simGetGroundTruthEnvironment()` and `simGetGroundTruthKinematics()`. In the case where you spawn two drones at separate locations and read their position immediately after spawning, you would expect their positions to be very similar since they wouldn't have moved very far from their respective origins.

I confirmed with a modified version of the [multi_agent_drone.py](https://github.com/microsoft/AirSim/blob/master/PythonClient/multirotor/multi_agent_drone.py) script that if you move two drones in different directions that they will then return different positions.

If you want to obtain a global position for the drone, you can make use of the `simGetObjectPose()` API, for example calling the following inside multi_agent_drone.py outputted the expected global position of each drone:

```
print(client.simGetObjectPose("Drone1").position)
print(client.simGetObjectPose("Drone2").position)
```

PR #4282 (2022-01-28): Update client.py to specify which APIs return positions in local frame
Add additional documentation specifying the frame of positions returned by various APIs.

Fixes: #4261    
Fixes: #3844
Fixes: #3882

## About
Add additional documentation specifying the frame of positions returned by various APIs.

## How Has This Been Tested?

## Screenshots (if appropriate):
closed 2022-01-28 by jonyMarino