microsoft/AirSim #4263: Object detection API gives incorrect detections

bozcani, 2022-01-03:
## Bug report 

- AirSim Version/#commit: v.1.6.0 Linux, 
- UE/Unity version: UE4.27 and UE4.25
- autopilot version: n/a
- OS Version: Ubuntu 20.04

### What's the issue you encountered?

AirSim [object detection API](https://microsoft.github.io/AirSim/object_detection/) gives incorrect bounding boxes for custom added objects. The detected bounding boxes flicker across the frames even if the drone is stationary. Here is the first screenshot. All objects are detected by Object Detection API but note that some detections still don't have tight bounding boxes as that cylinders have.

[image]

In the second image (just next frame), some detections disappear:
[image]

The detected bounding boxes appear and disappear randomly through frames. I tried with different settings (e.g., different flight paths, object types and sizes, object detection API parameters) but detection continues flickering. I don't think it is because of a processing power issue but it could be related with how object detection API is implemented. 

Related with https://github.com/microsoft/AirSim/issues/4196 but not the same.

### Settings

```
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "EngineAssociation": "4.27.1",
  "SimMode": "Multirotor",
  "ViewMode": "GroundObserver",
	"Recording": {
		"RecordOnMove": true,
		"RecordInterval": 0.20,
		"Enabled":false,
		"Cameras": [
			{ "VehicleName": "Drone0","CameraName": "0", "ImageType": 0, "PixelsAsFloat": false, "Compress": true },
			{ "VehicleName": "Drone0","CameraName": "0", "ImageType": 5, "PixelsAsFloat": false, "Compress": true }
		]
	},
  "CameraDefaults": {
      "CaptureSettings": [
        {
          "ImageType": 0,
          "Width": 1024,
          "Height": 576,
          "FOV_Degrees": 90,
          "AutoExposureSpeed": 100,
          "MotionBlurAmount": 0
        },
        {
          "ImageType": 5,
          "Width": 1024,
          "Height": 576,
          "FOV_Degrees": 90,
          "AutoExposureSpeed": 100,
          "MotionBlurAmount": 0
        }
    ],
    "Gimbal": {
      "Stabilization": 0
    }
  },
  "Vehicles": {
    "Drone0": {
      "VehicleType": "SimpleFlight",
      "X": 0, "Y": 0, "Z": 0, "Yaw": 270
    }
  }
}
```

### How can the issue be reproduced?

1. Add some objects in the `Blocks` environment.
2. Download and add your object meshes into detection tutorial file: [detection.py](https://github.com/microsoft/AirSim/blob/master/PythonClient/detection/detection.py)
3. Run detection.py.

### Include full error message in text form

**What's better than filing an issue? Filing a pull request :).**

rajat2004, 2022-01-03:
Not sure if you have already tried it, but can you increase the radius value using `simSetDetectionFilterRadius` to  a high value, say 1000m or what you feel is the max possible distance, to confirm it isn't due to the object being just at the edge

bozcani, 2022-01-04:
Yes, I tried a wide range of radius values. It doesn't fix. I can confirm that it is not related with being at the edge.

Zartris, 2022-01-04:
So I narrowed down the "problems":

First is the missing/flickering bounding boxes. 
This problem occurs in the [DetectionComponent.cpp#L136-L166](https://github.com/microsoft/AirSim/blob/bb9922eb0d56877d8e987ec1c24acc13c4ab3730/Unreal/Plugins/AirSim/Source/DetectionComponent.cpp#L136-L166), To check if it's actually visible or hidden, we check against 8 extend points by using LineTraceSingleChannel (meaning we are raytracing to see if we hit the object or if it is occluded). If the actor is in camera view but didn't hit any point out of 8 extended points, then check against 10 random points. This introduces some problems as the object might be considered "occluded" by mistake if it occupies a low number of pixels in the bounding box because a ray likely misses the target object (like t-shape object).
The solution to this... is still unknown to me, but if you really want the bounding boxes, you can disable the aforementioned lines and you will remove the occlusion check, but it will not be flickering or missing. (So this needs to be fixed in the future).

Secondly is the weird sized bounding boxes.
I tracked this down to only be a problem when working with the character class, and a solution to this is to change [DetectionComponent.cpp#L70](https://github.com/microsoft/AirSim/blob/bb9922eb0d56877d8e987ec1c24acc13c4ab3730/Unreal/Plugins/AirSim/Source/DetectionComponent.cpp#L70)
`actor->GetActorBounds(false, origin, extend);` to `actor->GetActorBounds(true, origin, extend);`
Hereby setting the OnlyCollidingComponents option to true. For reference:
```
void GetActorBounds
(
    bool bOnlyCollidingComponents,
    FVector & Origin,
    FVector & BoxExtent,
    bool bIncludeFromChildActors
) const 
```

The best way to explain this is to watch the video about [GetActorBounds](https://www.youtube.com/watch?v=Bat22Dn_9Qo) as he visually shows the box that we are hitting using the trace line. 

I'm not solving all your problems but I hope it helps for future reference.

bozcani, 2022-01-04:
Hey @Zartris, thanks for the help!

Setting `bOnlyCollidingComponents` to `true` seems to solve the whole problem. By setting this, the actor bounds computed by Unreal seem more correct. Here is when `bOnlyCollidingComponents=false` (current implementation of Airsim)
[image]
When we set `bOnlyCollidingComponents=true` 3D box seems tighter. 
[image]

When the 3D box is tighter, rays in the loops don't miss the object. So it seems there is no need to change the loops after setting `bOnlyCollidingComponents=true`. The detection looks much better now. I will test it with more scenes locally and possibly create a MR for the fix.
[image]

Gastastrophe, 2022-01-04:
If you want a temporary fix, I use the segmentation camera to create bounding boxes. You can then use the leftmost, bottommost, etc. coordinates in the image that contain the color corresponding to the object in question to create a bounding box. It's slow, but the accuracy is perfect.

bozcani, 2022-01-04:
Yes, it is a workaround solution that I use so far. However, I also need GPS data of the detected objects, and it is only possible if Object Detection API works properly.

closed 2022-01-04 by jonyMarino

ssaigarimella, 2025-07-24:
@Gastastrophe @bozcani by any chance do you how I would be able to get 3D bounding boxes directly?

Zartris, 2025-07-24:
Hi @ssaigarimella, its been years since I have used airsim but back when this was made there is a 3d tag in the return call you get from calling simGetDetections.

Example:
```python 
camera_name = "0"
image_type = airsim.ImageType.Scene

client = airsim.MultirotorClient()
client.confirmConnection()

client.simSetDetectionFilterRadius(camera_name, image_type, 80 * 100) # in [cm]
client.simAddDetectionFilterMeshName(camera_name, image_type, "Cylinder_*") 
client.simGetDetections(camera_name, image_type)
detections = client.simClearDetectionMeshNames(camera_name, image_type)
```
Detections will then contain
```python
Cylinder: <DetectionInfo> {   'box2D': <Box2D> {   'max': <Vector2r> {   'x_val': 617.025634765625,
    'y_val': 583.5487060546875},
    'min': <Vector2r> {   'x_val': 485.74359130859375,
    'y_val': 438.33465576171875}},
    'box3D': <Box3D> {   'max': <Vector3r> {   'x_val': 4.900000095367432,
    'y_val': 0.7999999523162842,
    'z_val': 0.5199999809265137},
    'min': <Vector3r> {   'x_val': 3.8999998569488525,
    'y_val': -0.19999998807907104,
    'z_val': 1.5199999809265137}},
    'geo_point': <GeoPoint> {   'altitude': 16.979999542236328,
    'latitude': 32.28772183970703,
    'longitude': 34.864785008379876},
    'name': 'Cylinder9_2',
    'relative_pose': <Pose> {   'orientation': <Quaternionr> {   'w_val': 0.9929741621017456,
    'x_val': 0.0038591264747083187,
    'y_val': -0.11333247274160385,
    'z_val': 0.03381215035915375},
    'position': <Vector3r> {   'x_val': 4.400000095367432,
    'y_val': 0.29999998211860657,
    'z_val': 1.0199999809265137}}}
```
Which contains the 3d box results.
This example is from the docs found here. So feel free to read more. https://microsoft.github.io/AirSim/object_detection/