microsoft/AirSim #4265: AirSim with Robot Dog?

szx0112, 2022-01-03:
## What feature are you suggesting?
### Overview:

Hi, is it possible to add a robot dog simulator in airsim? It's getting quite popular recently.

### Smaller Details: 

### Nature of Request:  

- Addition
- Change
- Removal
  - 

## Why would this feature be useful?

jonyMarino, 2022-01-06:
Hi and welcome @szx0112! It is not an easy task. You would need to make a lot of changes to the AirSim code. Changing the car's mesh to the one for a robot dog is simple, but I don't know if that suffices your needs.

szx0112, 2022-01-12:
@jonyMarino Thanks for the reply. 

I think the motion mechanism of the robot dog is more similar to a human (e.g., jump, turn, climb stairs) instead of a car (has the minimal turning radius). Is that possible to make such adjustment? 

The application I am interested in is the usage of robot dogs for indoor surveillance. Thanks.

Vampireqyz, 2022-01-13:
> 您好，欢迎@szx0112！这不是一件容易的事。您需要对 AirSim 代码进行大量更改。将汽车的网格更改为机器狗的网格很简单，但我不知道这是否满足您的需求。
[image]
[image]
when we set blocks for airsim,operated the "update_from_git.bat",there is no ".sln"files.thank you,looking forward to your reply.