microsoft/AirSim #4271: Attempting to get the camera information for an external camera crashes Unreal.

ab3nd, 2022-01-04:
AirSim Version/#commit: v1.6.0-windows
  UE/Unity version: 4.26
  autopilot version: N/A
  OS Version: Ubuntu 18.04.5 LTS (In WSL) and Windows 10 Pro

I have a settings.json that includes an external camera named "FixedCamera1", it's the same as the one in https://github.com/microsoft/AirSim/issues/4087 . When I attempt to get the camera information for that camera, the session of the Unreal Editor that I'm using to run AirSim crashes. 

Little interactive session to trigger the crash:
```
Python 3.6.9 (default, Jan 26 2021, 15:33:00)
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import airsim
>>> c = airsim.VehicleClient("10.42.0.16")
>>> c.simGetCameraInfo("FixedCamera1")
```

Beginning of Unreal's output once the crash is detected:
```
Unhandled Exception: EXCEPTION_ACCESS_VIOLATION reading address 0x0000000000000148

UE4Editor_AirSim!APIPCamera::getCameraInfo() [E:\Test_Environments\Plugins\AirSim\Source\PIPCamera.cpp:303]
UE4Editor_AirSim!<lambda_0d2c3b3216be71e77eadf656d901d44b>::operator()() [E:\Test_Environments\Plugins\AirSim\Source\WorldSimApi.cpp:739]
UE4Editor_AirSim!TGraphTask<TFunctionGraphTaskImpl<void __cdecl(void),0> >::ExecuteTask() [E:\UE_4.26\Engine\Source\Runtime\Core\Public\Async\TaskGraphInterfaces.h:886]
UE4Editor_Core
UE4Editor_Core
```

ab3nd, 2022-01-04:
In the file that it mentions, PIPCamera.cpp:
```
msr::airlib::CameraInfo APIPCamera::getCameraInfo() const
{
    msr::airlib::CameraInfo camera_info;

    camera_info.pose.position = ned_transform_->toLocalNed(this->GetActorLocation());
    camera_info.pose.orientation = ned_transform_->toNed(this->GetActorRotation().Quaternion());
    camera_info.fov = camera_->FieldOfView;
    camera_info.proj_mat = getProjectionMatrix(ImageType::Scene);
    return camera_info;
}
```
Line 303 is the line to get the position of the camera.

ab3nd, 2022-01-04:
Found it. 

c.simGetCameraInfo("FixedCamera1") -> SUDDEN DEATH
c.simGetCameraInfo("FixedCamera1", external=True) -> Information, which is indeed about cameras

"External" is VERY not optional.

closed 2022-01-04 by ab3nd

sandilyasg, 2024-01-12:
@ab3nd external field is not supported on Unity Image APIs. When I run simGetCameraInfo it crashes, do you have any experience with solving this?