microsoft/AirSim #4276: Azure example fails to run

rhalaly, 2022-01-08:
## Bug report 

- AirSim Version/#commit: bb9922eb0d56877d8e987ec1c24acc13c4ab3730
- UE/Unity version: 4.27
- autopilot version:
- OS Version: Windows 11

### What's the issue you encountered?

Running the [Azure example](https://github.com/microsoft/AirSim/tree/master/azure) with docker (built with `docker build -t <your-registry-url>/<your-image-name> -f ./docker/Dockerfile .` from [docs](https://microsoft.github.io/AirSim/azure/#building-the-docker-image)) fails to run.

<details>
<summary>Logs</summary>

```
LogInit: Compatible Engine Version: 4.24.0-0+++UE4+Release-4.24
LogInit: Net CL: 0
LogInit: OS: GenericOSVersionLabel (GenericOSSubVersionLabel), CPU: AMD Ryzen 9 5900X 12-Core Processor            , GPU: GenericGPUBrand
LogInit: Compiled (64-bit): Feb  3 2020 21:23:28
LogInit: Compiled with Clang: 8.0.1 (tags/RELEASE_801/final)
LogInit: Build Configuration: Development
LogInit: Branch Name: ++UE4+Release-4.24
LogInit: Command Line:
LogInit: Base Directory: /home/airsim_user/Blocks/Blocks/Binaries/Linux/
LogInit: Installed Engine Build: 0
LogDevObjectVersion: Number of dev versions registered: 25
LogDevObjectVersion:   Dev-Blueprints (B0D832E4-1F89-4F0D-ACCF-7EB736FD4AA2): 10
LogDevObjectVersion:   Dev-Build (E1C64328-A22C-4D53-A36C-8E866417BD8C): 0
LogDevObjectVersion:   Dev-Core (375EC13C-06E4-48FB-B500-84F0262A717E): 3
LogDevObjectVersion:   Dev-Editor (E4B068ED-F494-42E9-A231-DA0B2E46BB41): 37
LogDevObjectVersion:   Dev-Framework (CFFC743F-43B0-4480-9391-14DF171D2073): 36
LogDevObjectVersion:   Dev-Mobile (B02B49B5-BB20-44E9-A304-32B752E40360): 2
LogDevObjectVersion:   Dev-Networking (A4E4105C-59A1-49B5-A7C5-40C4547EDFEE): 0
LogDevObjectVersion:   Dev-Online (39C831C9-5AE6-47DC-9A44-9C173E1C8E7C): 0
LogDevObjectVersion:   Dev-Physics (78F01B33-EBEA-4F98-B9B4-84EACCB95AA2): 2
LogDevObjectVersion:   Dev-Platform (6631380F-2D4D-43E0-8009-CF276956A95A): 0
LogDevObjectVersion:   Dev-Rendering (12F88B9F-8875-4AFC-A67C-D90C383ABD29): 36
LogDevObjectVersion:   Dev-Sequencer (7B5AE74C-D270-4C10-A958-57980B212A5A): 11
LogDevObjectVersion:   Dev-VR (D7296918-1DD6-4BDD-9DE2-64A83CC13884): 2
LogDevObjectVersion:   Dev-LoadTimes (C2A15278-BFE7-4AFE-6C17-90FF531DF755): 1
LogDevObjectVersion:   Private-Geometry (6EACA3D4-40EC-4CC1-B786-8BED09428FC5): 3
LogDevObjectVersion:   Dev-AnimPhys (29E575DD-E0A3-4627-9D10-D276232CDCEA): 17
LogDevObjectVersion:   Dev-Anim (AF43A65D-7FD3-4947-9873-3E8ED9C1BB05): 2
LogDevObjectVersion:   Dev-ReflectionCapture (6B266CEC-1EC7-4B8F-A30B-E4D90942FC07): 1
LogDevObjectVersion:   Dev-Automation (0DF73D61-A23F-47EA-B727-89E90C41499A): 1
LogDevObjectVersion:   FortniteMain (601D1886-AC64-4F84-AA16-D3DE0DEAC7D6): 31
LogDevObjectVersion:   Dev-Enterprise (9DFFBCD6-494F-0158-E221-12823C92A888): 8
LogDevObjectVersion:   Dev-Niagara (F2AED0AC-9AFE-416F-8664-AA7FFA26D6FC): 1
LogDevObjectVersion:   Dev-Destruction (174F1F0B-B4C6-45A5-B13F-2EE8D0FB917D): 9
LogDevObjectVersion:   Dev-Physics-Ext (35F94A83-E258-406C-A318-09F59610247C): 8
LogDevObjectVersion:   Dev-CineCamera (B2E18506-4273-CFC2-A54E-F4BB758BBA07): 1
LogInit: Presizing for max 2097152 objects, including 1 objects not considered by GC, pre-allocating 0 bytes for permanent pool.
LogStreaming: Display: Async Loading initialized: Event Driven Loader: true, Async Loading Thread: false, Async Post Load: true
LogInit: Object subsystem initialized
LogConfig: Setting CVar [[con.DebugEarlyDefault:1]]
LogConfig: Setting CVar [[r.setres:1280x720]]
[2022.01.08-16.52.55:949][  0]LogConfig: Setting CVar [[con.DebugEarlyDefault:1]]
[2022.01.08-16.52.55:949][  0]LogConfig: Setting CVar [[r.setres:1280x720]]
[2022.01.08-16.52.55:949][  0]LogConfig: Applying CVar settings from Section [/Script/Engine.RendererSettings] File [../../../Blocks/Saved/Config/LinuxNoEditor/Engine.ini]
[2022.01.08-16.52.55:949][  0]LogConfig: Setting CVar [[r.GPUCrashDebugging:0]]
[2022.01.08-16.52.55:949][  0]LogConfig: Applying CVar settings from Section [/Script/Engine.RendererOverrideSettings] File [../../../Blocks/Saved/Config/LinuxNoEditor/Engine.ini]
[2022.01.08-16.52.55:949][  0]LogConfig: Applying CVar settings from Section [/Script/Engine.StreamingSettings] File [../../../Blocks/Saved/Config/LinuxNoEditor/Engine.ini]
[2022.01.08-16.52.55:949][  0]LogConfig: Setting CVar [[s.MinBulkDataSizeForAsyncLoading:131072]]
[2022.01.08-16.52.55:949][  0]LogConfig: Setting CVar [[s.AsyncLoadingThreadEnabled:0]]
[2022.01.08-16.52.55:949][  0]LogConfig: Setting CVar [[s.EventDrivenLoaderEnabled:1]]
[2022.01.08-16.52.55:949][  0]LogConfig: Setting CVar [[s.WarnIfTimeLimitExceeded:0]]
[2022.01.08-16.52.55:949][  0]LogConfig: Setting CVar [[s.TimeLimitExceededMultiplier:1.5]]     
[2022.01.08-16.52.55:949][  0]LogConfig: Setting CVar [[s.TimeLimitExceededMinTime:0.005]]      
[2022.01.08-16.52.55:949][  0]LogConfig: Setting CVar [[s.UseBackgroundLevelStreaming:1]]       
[2022.01.08-16.52.55:949][  0]LogConfig: Setting CVar [[s.PriorityAsyncLoadingExtraTime:15.0]]  
[... 748 lines cut ...]
    if self.ping():
  File "/home/airsim_user/.local/lib/python3.6/site-packages/airsim/client.py", line 35, in ping
    return self.client.call('ping')
  File "/home/airsim_user/.local/lib/python3.6/site-packages/msgpackrpc/session.py", line 41, in call
    return self.send_request(method, args).get()
  File "/home/airsim_user/.local/lib/python3.6/site-packages/msgpackrpc/future.py", line 43, in get
    raise self._error
msgpackrpc.error.TransportError: Retry connection over the limit
```

</details>

### Settings
Default from Blocks environment.

### How can the issue be reproduced?

1. Follow these [docs](https://microsoft.github.io/AirSim/azure/#building-the-docker-image)
2. Clone AirSim repo
3. Run `cd azure`
4. Run `docker build -t <your-registry-url>/<your-image-name> -f ./docker/Dockerfile .`
5. Run `docker run <your-registry-url>/<your-image-name>`

### Include full error message in text form
See logs above.

jonyMarino, 2022-01-31:
Hi @rhalaly! That docker image was only created to be used on the azure server. The docker images we support are the ones created by the scripts in the docker folder as detailed in the documentation.

rhalaly, 2022-01-31:
Hi @jonyMarino ! Yes I know, I saw the same logs in an NV12 machine on Azure