microsoft/AirSim #4280: cd Airsim

Vampireqyz, 2022-01-12:
[image]

alonfaraj, 2022-01-12:
Seems like you running on windows so you can't run the build batch file from git bash.
You need to open `Visual Studio 2019 Developer Command Prompt`.
Then cd into AirSim root directory and run `build.cmd`.

Vampireqyz, 2022-01-13:
> 好像你在 Windows 上运行，所以你不能从 git bash 运行构建批处理文件。 你需要打开`Visual Studio 2019 Developer Command Prompt`. 然后 cd 进入 AirSim 根目录并运行`build.cmd`.
[image]
DO you know how to solve this ,when we set blocks for airsim ,operate "update_from_git.bat",there is no ".sln" files.looking forward to your reply.

alonfaraj, 2022-01-13:
you need to run `GenerateProjectFiles.bat` before

Vampireqyz, 2022-01-13:
> 你需要运行`GenerateProjectFiles.bat`之前

It's not work, it seems like we can't find "hkey_classes_root\unreal.projectfile\shell\rungenproj"
[image]

Vampireqyz, 2022-01-13:
> 你需要运行`GenerateProjectFiles.bat`之前

when we double click "Block.sln" file,operate this file which shows can not import "delayimp.lib"file.
[image]
[image]

jonyMarino, 2022-01-31:
Hi @Vampireqyz! What version of vs are you using? Try to follow our documentation.

zimmy87, 2022-01-31:
Hi @Vampireqyz, from looking [here](https://stackoverflow.com/questions/26410604/cannot-open-input-file-delayimp-lib), it looks like delayimp.lib is a part of MSVC. I'm using `MSVC v142 - VS 2019 C++ x64/x86 build tools` locally. Make sure you have that component installed and try again.

[image]