microsoft/AirSim #4281: Blank screen, nothing visible until ALT+F4

aib, 2022-01-12:
## Bug report 

- AirSim Version/#commit: releases/tag/v1.6.0-linux
- UE/Unity version: Unknown
- autopilot version: N/A
- OS Version: Ubuntu 21.20

### What's the issue you encountered?

Upon launching LinuxNoEditor/Blocks.sh or LinuxNoEditor/Blocks/Binaries/Linux/Blocks I get a black screen with my mouse cursor visible. From the alt-tab dialog I can surmise that a "Choose Vehicle" window and a simulation window have been created. At this point, the desktop the simulation is running at is blocked, I cannot see or visibly switch to any windows on the desktop, including the console that ran the program. If I focus the Choose Vehicle window and close it, the simulation window becomes normal and I am able to use everything normally, including the simulation.

Is the Choose Vehicle window a modal dialog over a fullscreen application which takes over the screen?

Where is the windowed mode setting? `--help` produces nothing and the documentation only refers to simulation configuration JSON files.

### Settings

No settings. I would play with them if I could find them.

Did a search for "fullscreen" and converted a 1 to 0 and a True to False and the only change is that I see a quarter-screen window for a split second before I get the working fullscreen simulation window, after fumblingly closing Choose Vehicle.

### How can the issue be reproduced?

1. Download and extract the release
2. Run it

### Include full error message in text form

N/A

**What's better than filing an issue? Filing a pull request :).**

Well, this is my first introduction to this program, or any of its kind. This is a computer/setup that runs hundreds of games through Steam and Proton.

aib, 2022-01-12:
stdout:
```
Increasing per-process limit of core file size to infinity.
LogPlatformFile: Using cached read wrapper
LogTaskGraph: Started task graph with 5 named threads and 17 total threads with 3 sets of task threads.
LogStats: Stats thread started at 0.026031
LogICUInternationalization: ICU TimeZone Detection - Raw Offset: +3:00, Platform Override: ''
LogPluginManager: Mounting plugin LightPropagationVolume
LogPluginManager: Mounting plugin AISupport
LogPluginManager: Mounting plugin LauncherChunkInstaller
LogPluginManager: Mounting plugin Niagara
LogPluginManager: Mounting plugin ImgMedia
LogPluginManager: Mounting plugin WmfMedia
LogPluginManager: Mounting plugin AvfMedia
LogPluginManager: Mounting plugin MediaCompositing
LogPluginManager: Mounting plugin WebMMedia
LogPluginManager: Mounting plugin NetcodeUnitTest
LogPluginManager: Mounting plugin ChaosEditor
LogPluginManager: Mounting plugin PlatformCrypto
LogPluginManager: Mounting plugin GeometryProcessing
LogPluginManager: Mounting plugin ChaosCloth
LogPluginManager: Mounting plugin AutomationUtils
LogPluginManager: Mounting plugin GeometryCache
LogPluginManager: Mounting plugin CharacterAI
LogPluginManager: Mounting plugin BackChannel
LogPluginManager: Mounting plugin PlanarCut
LogPluginManager: Mounting plugin ChaosSolverPlugin
LogPluginManager: Mounting plugin AlembicImporter
LogPluginManager: Mounting plugin GeometryCollectionPlugin
LogPluginManager: Mounting plugin ChaosNiagara
LogPluginManager: Mounting plugin ChaosClothEditor
LogPluginManager: Mounting plugin UObjectPlugin
LogPluginManager: Mounting plugin AnimationSharing
LogPluginManager: Mounting plugin DatasmithContent
LogPluginManager: Mounting plugin VariantManagerContent
LogPluginManager: Mounting plugin TcpMessaging
LogPluginManager: Mounting plugin UdpMessaging
LogPluginManager: Mounting plugin MeshPainting
LogPluginManager: Mounting plugin GeometryMode
LogPluginManager: Mounting plugin AssetManagerEditor
LogPluginManager: Mounting plugin EditorScriptingUtilities
LogPluginManager: Mounting plugin FacialAnimation
LogPluginManager: Mounting plugin OnlineSubsystem
LogPluginManager: Mounting plugin OnlineSubsystemUtils
LogPluginManager: Mounting plugin OnlineSubsystemNull
LogPluginManager: Mounting plugin ScreenshotTools
LogPluginManager: Mounting plugin Paper2D
LogPluginManager: Mounting plugin AudioCapture
LogPluginManager: Mounting plugin PhysXVehicles
LogPluginManager: Mounting plugin ExampleDeviceProfileSelector
LogPluginManager: Mounting plugin GooglePAD
LogPluginManager: Mounting plugin AssetTags
LogPluginManager: Mounting plugin Synthesis
LogPluginManager: Mounting plugin LinuxDeviceProfileSelector
LogPluginManager: Mounting plugin CustomMeshComponent
LogPluginManager: Mounting plugin LocationServicesBPLibrary
LogPluginManager: Mounting plugin SignificanceManager
LogPluginManager: Mounting plugin ArchVisCharacter
LogPluginManager: Mounting plugin AppleImageUtils
LogPluginManager: Mounting plugin SoundFields
LogPluginManager: Mounting plugin WebMMoviePlayer
LogPluginManager: Mounting plugin CableComponent
LogPluginManager: Mounting plugin RuntimePhysXCooking
LogPluginManager: Mounting plugin EditableMesh
LogPluginManager: Mounting plugin AndroidPermission
LogPluginManager: Mounting plugin MobilePatchingUtils
LogPluginManager: Mounting plugin ProceduralMeshComponent
LogPluginManager: Mounting plugin MatineeToLevelSequence
LogPluginManager: Mounting plugin TemplateSequence
LogPluginManager: Mounting plugin MovieRenderPipeline
[... 933 lines cut ...]
[2022.01.12-08.36.43:867][310]LogModuleManager: Shutting down and abandoning module Renderer (18)
[2022.01.12-08.36.43:867][310]LogModuleManager: Shutting down and abandoning module Engine (16)
[2022.01.12-08.36.43:867][310]LogModuleManager: Shutting down and abandoning module CoreUObject (14)
[2022.01.12-08.36.43:867][310]LogModuleManager: Shutting down and abandoning module NetworkFile (12)
[2022.01.12-08.36.43:867][310]LogModuleManager: Shutting down and abandoning module CookedIterativeFile (10)
[2022.01.12-08.36.43:867][310]LogModuleManager: Shutting down and abandoning module StreamingFile (8)
[2022.01.12-08.36.43:867][310]LogModuleManager: Shutting down and abandoning module SandboxFile (6)
[2022.01.12-08.36.43:867][310]LogModuleManager: Shutting down and abandoning module PakFile (4)
[2022.01.12-08.36.43:867][310]LogModuleManager: Shutting down and abandoning module RSA (3)
[2022.01.12-08.36.43:867][310]LogContentStreaming: Display: There are 1 unreleased StreamingManagers
[2022.01.12-08.36.43:898][310]LogExit: Exiting.
[2022.01.12-08.36.43:898][310]LogInit: Tearing down SDL.
```

stderr:
```
Ensure condition failed: VendorId != EGpuVendorId::Unknown [File:/home/ue4/ue_build/Engine/Source/Runtime/VulkanRHI/Private/VulkanDevice.cpp] [Line: 182] 

XYXYXY XYXYXY Clearnupworld 0x7f4ec4e45160 -> (nil)
```
(this is where I close Choose Vehicle)
```
Resizing viewport due to setres change, 1920 x 1080
XYXYXY XYXYXY Clearnupworld 0x7f4ec4e40200 -> (nil)
```

aib, 2022-01-12:
Just found out about `-windowed` thorough an external site. Fixes the problem as expected.

jonyMarino, 2022-01-31:
Thanks, @aib, for solving by yourself, and keep us informed!

closed 2022-01-31 by jonyMarino