microsoft/AirSim #4285: Unity Simulator Crashes When Using External Physics Engine Setting

ClarkZha, 2022-01-13:
AirSim Version/#commit: 1.7.0
UE/Unity version: Unity 2019.3.12.f1
autopilot version: - N/A
OS Version: Ubuntu 18.04

## Bug report 

Unity simulation will crash if ` "PhysicsEngineName":"ExternalPhysicsEngine"` is set in the settings.json. 

The Unity editor will crash once the drone simulator is chose in the scene selection.
[image]

A quick dig in the code base suggests that the external physics engine option may not be setup properly in the Unity wrapper.
For example, in SimWorldModeBase.cpp for Unity wrapper, the "ExternalPhysicsEngine" will not be parsed and an error will be thrown. 
[image]

jonyMarino, 2022-01-31:
Hi @ClarkZha! you can fix it adding these lines before the last else:
```c++
    else if (physics_engine_name == "ExternalPhysicsEngine") {        
        physics_engine.reset(new msr::airlib::ExternalPhysicsEngine());
    }
```

ClarkZha, 2022-01-31:
Hi @jonyMarino! Thanks for the response. 

I did a similar hack earlier and it seemed to work. I wasn't sure if this would cause bug in other places in the code so I didn't post it here to avoid distract fellow developers. Glad this was the right approach : )

closed 2022-01-31 by ClarkZha